PiSupply/PaPiRus #3: epd init script not working on Raspberry Pi2

mad61, 2016-01-14:
The epd init script is not working on RPi2 (running Raspian Jessy).

After changing line 34 in epd-fuse:

34c34
<     modprobe spi-bcm2708
---
>     modprobe spi-bcm2835

everything works.

shawaj, 2016-01-14:
We are aware of this issue and will be fixing it as soon as possible here.

Have you found any other errors with any of the Python scripts or anything?

mad61, 2016-01-14:
I fixed the following yet.

##### papirus/text.py
```patch
diff --git a/papirus/text.py b/papirus/text.py
index a04c84f..e86a7ed 100644
--- a/papirus/text.py
+++ b/papirus/text.py
@@ -34,9 +34,9 @@ class PapirusText():
                 text_lines[current_line] += " " + word
             else:
                 # No space left on line so move to next one
-    	    text_lines.append("")
+    	        text_lines.append("")
                 current_line += 1
-    	    text_lines[current_line] += " " + word
+    	        text_lines[current_line] += " " + word
 
         current_line = 0
         for l in text_lines:
```
line 37 and 39 needs additional indention

##### Sample Basic API
```patch
diff --git a/README.md b/README.md
index df9f667..ae0bc51 100644
--- a/README.md
+++ b/README.md
@@ -48,12 +48,13 @@ sudo service epd-fuse start
 
 ```python
 from papirus import Papirus
+from PIL import Image
 
 # The epaper screen object
 screen = Papirus()
 
 # Write a bitmap to the epaper screen
-screen.display('./path/to/bmp/image')
+screen.display(Image.open('./path/to/bmp/image').resize(screen.size))
 
 # Perform a full update to the screen (slower)
 screen.update()
@@ -63,7 +64,7 @@ screen.partial_update()
 
 # Change screen size
 # SCREEN SIZES 1_44INCH | 1_9INCH | 2_0INCH | 2_6INCH | 2_7INCH
-screen.set_size(papirus.2_7INCH)
+# screen.set_size(papirus.2_7INCH)
 
```
"screen.display('./path/to/bmp/image')" and set_size not working. The "display" method needs a Image object not a string. "set_size" I didn't debug further.

##### bin/papirus-write font issue:
```patch
diff --git a/bin/papirus-write b/bin/papirus-write
index e857353..ff8fb63 100644
--- a/bin/papirus-write
+++ b/bin/papirus-write
@@ -42,7 +42,7 @@ def write_text(papirus, text, size):
     # prepare for drawing
     draw = ImageDraw.Draw(image)
 
-    font = ImageFont.truetype('Pillow/Tests/fonts/FreeMono.ttf', size)
+    font = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf', size)
 
     # Calculate the max number of char to fit on line
     line_size = (papirus.width / (size*0.65))
```

On Jessie Lite a font package
```bash
apt-get install fonts-freefont-ttf
```
needs to be installed before.

shawaj, 2016-01-16:
Could you try again now?

delormev, 2016-01-17:
@mad61 is correct re: spi-bcm2835 on a Pi 2, and also in the modifications he had to make to the different python scripts! I've made the same changes and most things are working correctly now!

shawaj, 2016-01-17:
@delormev @mad61 could you try again with the updated code?

delormev, 2016-01-17:
@shawaj I don't think the code had been updated... The issues were still there when I checked out the code a couple of hours ago (see for example line 37 and 39 of https://github.com/PiSupply/PaPiRus/blob/master/papirus/text.py)
So was the issue with edp-fuse (but I guess that's not on your side, I've also opened an issue for @repaper/gratis)

shawaj, 2016-01-17:
@delormev - hmmm thought we had fixed all of the issues. Will take another look now. The epd-fuse one is the spi-bcm2835 one right? I can email straight to the relevant person at repaper too.

Sheffyp, 2016-01-19:
Hey guys - I'm getting pretty much all the same errors as above. Im running a B+ with the latest wheezy image. I can't resolve the indent issue even though I've dragged the text.py file into every editor I have. Do you have any pointers for a new install?

sampsakuronen, 2016-01-23:
I have spent an hour debugging this. I'm running the newest Raspbian Jessie.

Installing `libfuse-dev` fails, this `.deb` is not found. Version `deb8u2` is available here: http://mirrordirector.raspbian.org/raspbian/pool/main/f/fuse/

    E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/f/fuse/libfuse-dev_2.9.3-15+deb8u1_armhf.deb  404  Not Found [IP: 5.153.225.207 80]

After fixing the intendation and font path bugs in scripts I'm at this point:

    pi@raspberrypi:/tmp $ sudo papirus-write "foo-bar"
    Traceback (most recent call last):
      File "/usr/local/bin/papirus-write", line 67, in <module>
        main()
      File "/usr/local/bin/papirus-write", line 23, in main
        papirus = Papirus()
      File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 66, in __init__
        with open(os.path.join(self._epd_path, 'version')) as f:
    IOError: [Errno 107] Transport endpoint is not connected: '/dev/epd/version'

    pi@raspberrypi:/tmp $ sudo papirus-clock
    Traceback (most recent call last):
      File "/usr/local/bin/papirus-clock", line 26, in <module>
        from EPD import EPD
    ImportError: No module named EPD

jayeshjoy, 2016-01-25:
@sampsakuronen:
The no module named EPD error is because of an error in the script. There's another PR out to fix that.
I was getting a 404 with the fuse install till I did a apt-get update.

gaerank, 2016-01-25:
I have same issue with sampsakuronen
RPi2 
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"

pi@raspberrypi:~ $ sudo papirus-write "Hello world!"
Traceback (most recent call last):
  File "/usr/local/bin/papirus-write", line 67, in <module>
    main()
  File "/usr/local/bin/papirus-write", line 23, in main
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 66, in __init__
    with open(os.path.join(self._epd_path, 'version')) as f:
IOError: [Errno 2] No such file or directory: '/dev/epd/version'

pi@raspberrypi:~ $ sudo papirus-clock
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 26, in <module>
    from EPD import EPD
ImportError: No module named EPD
pi@raspberrypi:~ $

jayeshjoy, 2016-01-26:
@gaerank : For the "No module named EPD" error, try the changes listed here: https://github.com/PiSupply/PaPiRus/pull/5/files
Or a simpler one line change is to change line 26 in the papiru-clock script from
```
from EPD import EPD
```
to 
```
from papirus import Papirus as EPD
```

For the "No such file or directory: '/dev/epd/version'" error, try running 'sudo papirus-config' -> 'Enable Driver' -> 'Disable'

gaerank, 2016-01-26:
Hi,
First, I modified three files as your guide.
I did the papirus-config enable->disable driver. 
yeah, the "No such file or directory..." message has gone.
But I have still other errors.

Here is what I am done.
1. sudo papirus-config  select display size -> 2.7
2. ls -al /dev/epd   <<< can see fines
3. sudo papirus-config -> enable driver. 
4. ls -al /dev/epd   <<< empty
5. sudo papirus-config -> disable driver. 
6. ls -al /dev/epd   <<< can see fines
7. sudo papirus-clock
8. ls -al /dev/epd        ls -al /dev/      <<< get message "ls: cannot access epd: Transport endpoint is not connected"

hope this help to find problems.

Thank you.

\#1
pi@raspberrypi:~ $ sudo papirus-config
Your chosen option: Screen Size
2.7
2.7

\#2
pi@raspberrypi:~ $ ls -al /dev/epd
total 0
drwxrwxrwt  2 root root    0 Jan  1  1970 .
drwxr-xr-x 15 root root 3420 Jan 26 21:11 ..
drwxrwxrwx  2 root root    0 Jan  1  1970 BE
drwxrwxrwx  2 root root    0 Jan  1  1970 LE
--w--w--w-  1 root root    1 Jan  1  1970 command
-r--r--r--  1 root root 5808 Jan  1  1970 current
-r--r--r--  1 root root 5808 Jan  1  1970 current_inverse
-rw-rw-rw-  1 root root 5808 Jan  1  1970 display
-rw-rw-rw-  1 root root 5808 Jan  1  1970 display_inverse
-r--r--r--  1 root root   31 Jan  1  1970 panel
-rw-rw-rw-  1 root root    4 Jan  1  1970 temperature
-r--r--r--  1 root root    2 Jan  1  1970 version

\#3
pi@raspberrypi:~ $ sudo papirus-config  
Your chosen option: Enable Driver
insserv: warning: current start runlevel(s) (empty) of script `epd-fuse' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `epd-fuse' overrides LSB defaults (0 1 6).

\#4
pi@raspberrypi:~ $ ls -al /dev/epd
total 0
drwxr-xr-x  2 root root   40 Jan 26 21:11 .
drwxr-xr-x 15 root root 3420 Jan 26 21:11 ..
pi@raspberrypi:~ $ 

\#5
pi@raspberrypi:~ $ sudo papirus-config
Your chosen option: Enable Driver

\#6
pi@raspberrypi:~ $ ls -al /dev/epd
total 0
drwxrwxrwt  2 root root    0 Jan  1  1970 .
drwxr-xr-x 15 root root 3420 Jan 26 21:11 ..
drwxrwxrwx  2 root root    0 Jan  1  1970 BE
drwxrwxrwx  2 root root    0 Jan  1  1970 LE
--w--w--w-  1 root root    1 Jan  1  1970 command
-r--r--r--  1 root root 5808 Jan  1  1970 current
-r--r--r--  1 root root 5808 Jan  1  1970 current_inverse
-rw-rw-rw-  1 root root 5808 Jan  1  1970 display
-rw-rw-rw-  1 root root 5808 Jan  1  1970 display_inverse
-r--r--r--  1 root root   31 Jan  1  1970 panel
-rw-rw-rw-  1 root root    4 Jan  1  1970 temperature
-r--r--r--  1 root root    2 Jan  1  1970 version

\#7
pi@raspberrypi:~ $ sudo papirus-clock
panel = EPD 2.7 264 x 176  version=4 COG=2 FILM=231
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 121, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 68, in main
    papirus.clear()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 152, in clear
    self._command('C')
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 156, in _command
    f.write(c)
IOError: [Errno 103] Software caused connection abort

\#8
pi@raspberrypi:~ $ ls -al /dev/epd
ls: cannot access /dev/epd: Transport endpoint is not connected

pi@raspberrypi:~ $ ls -al /dev/
ls: cannot access epd: Transport endpoint is not connected
--snip--
drwxr-xr-x  6 root root       120 Jan 25 23:08 disk
d?????????  ? ?    ?            ?            ? epd
crw-rw----  1 root video  29,   0 Jan 25 23:08 fb0
--snip--

sampsakuronen, 2016-01-30:
Please see this tutorial. I got rid of the errors, but I seem to have a broken display, so I can't get anything to show up on the screen.

http://frederickvandenbosch.be/?p=1483

gaerank, 2016-01-30:
Hi,
Thank you for the information.
First time I got the package, I have no idea with the pogo pin. Now I know
what pogo pin for.
So, I solder it as the link page. same step as the page but the results is
same.

I don't know what's wrong on my PaPirus.

Thank you
Best Regards,

Kevin.

gaerank, 2016-01-30:
Hello,
I found something wrong.  I didn't touch the button battery when I got the package. 
The button battery was not correctly assembled.
As you know the battery is pre-assembled.

[image]

When I remove the battery, I found the -(minus) pin got bend as below picture.
[image]

Could this make my papirus defect?

Kevin.

YoungAtHome, 2016-01-31:
Continuing to get ImportError: No module named epd
File __init__.py
From EPD import EPD as Papiris

When I use papirus-config to enable the driver I get an error. But it seems to be enabled on startup. Evidence is files in /dev/epd

Any suggestions welcome.

vogonsaudi, 2016-01-31:
YoungAtHome, I followed these instructions http://frederickvandenbosch.be/?p=1483 and also created a simple, thrown together wiki up there ^^^ to show the steps I had to take to get it working.

To fix your problem, change
#from EPD import EPD

to

from papirus import Papirus as EPD

stephane-wis, 2016-01-31:
[image]

gaerank, 2016-02-02:
Hi,
I followed  http://frederickvandenbosch.be/?p=1483 but same error.

pi@raspberrypi:/ $ sudo papirus-clock 
panel = EPD 2.7 264 x 176  version=4 COG=2 FILM=231
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 122, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 69, in main
    epd.clear()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 152, in clear
    self._command('C')
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 156, in _command
    f.write(c)
IOError: [Errno 103] Software caused connection abort

pi@raspberrypi:/ $ ls /dev/epd/
ls: cannot access /dev/epd/: Transport endpoint is not connected
pi@raspberrypi:/ $

YoungAtHome, 2016-02-02:
Transport error means no SPI so neec to enable in raspi-config and reboot.

I have papirus working with python 2.7 but not 3.2.

Nick

gaerank, 2016-02-02:
Hi YoungAtHome,
I ran raspi-config for enable SPI.

pi@raspberrypi:~ $ sudo raspi-config
9 Advanced Options  > A6 SPI  > Would you like the SPI interface to be enabled? > Yes > Yes > Yes> Yes> Finish > Reboot

It is working.
Thank you YoungAtHome for sharing.

Here is my conclusion.

1. enable SPI by run raspi-config   <<< mandatory!!!
2. Reboot
3. modify /usr/local/bin/papirus-clock
  - from EPD import EPD 
  + from papirus import Papirus as EPD
4. modify /usr/local/bin/papirus-write 
  -    font = ImageFont.truetype('Pillow/Tests/fonts/FreeMono.ttf', size)
 +    font = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf', size)
add tab below two lines each head.
        text_lines.append("")
        text_lines[current_line] += " " + word
5.  Not sure to remove python3.  I've tested remove python3 and re-installed but the python3 was not a problem.

Best Regards,

Kevin.

stephane-wis, 2016-02-02:
Very happy to read that it’s working for you ;-)

YoungAtHome, 2016-02-04:
I was continuing to get the indent error in text.py after changing the code, so I cleared the python3 cache files for papirus
`cd /usr/local/lib/python3/dist-packages/papirus`
`sudo find . -name '*.pyc' -delete`

Rerun the clock to test.
`sudo papirus-clock`

Regards,
Nick

scargill, 2016-02-04:
I am getting that indent error - after fixing other issues.. so right now - the clock works but keeps going backwards - and the text - I keep getting an indent control where there is no faulty indent.  Can the organisation please fix the stuff out there so that instead of having to sail through forums we can just run the product and not have to get into clearing caches etc....  not really good enough though I and others are grateful to those who've taken the time to fix this stuff - we really should not have to be doing this for something as basic as getting text on a display.

trashcan, 2016-02-10:
```
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 136, in display
    raise EPDError('image size mismatch')
papirus.epd.EPDError: 'image size mismatch'
```
What resolution should I use for the "large" display?

YoungAtHome, 2016-02-10:
Display sizes are:
Small
1.44" diagonal and 128 x 96 pixel resolution
Medium
 2.0" diagonal and 200 x 96 pixel resolution
Large
2.7" diagonal and 264 x 176 pixel resolution

As I recall there may be a default of 2.0 (medium) in the epd.py script.

Use
sudo papirus-set 2.7
or
sudo papirus-config
to set.

Warning: Enable and and disable driver may be swapped in config so choose
and read on screen to check.

Good luck.

I have mine working on Python3.

Nick

deadsource, 2016-02-16:
It works so well that the whole screen went black just from running `papirus-clock`.

Followed all of the fixes metioned above till it started working and then after around 20 seconds the screen filled itself for some reason. `papirus-clear` doesn't clear the screen either.
[image]
I tried running `papirus-gol` afterwards and the screen cleared a bit (if you can see it in the image above).

Also, why is the pogo pin needed?

YoungAtHome, 2016-02-16:
Congratulations.

The pogo pin is a gpio connector to the reset point on the Pi so setting
the papirus end of the pin triggers either a reset or start (if Pi is off).
Gives wake-on-alarm but no clear documentation.

I have also got a black screen that doesn't clear.
Going to try on another hat but this was my large screen.
I have written code that tried to change the screen too fast and that left
it black. It may be overclocking or fast pi2 that are not expected by
papirus.

Need to get input from Aaron and co.

Regards,
Nick

deadsource, 2016-02-20:
Any luck getting the screen to clear? I have noticed that if I run `papirus-clock` long enough and then exit it, the pixels that are left from the last digit of the seconds appear as white on the yet to reset black screen. Though during all that time running the program, the screen remains black.

YoungAtHome, 2016-02-20:
That's interesting. I've still not cleared my large screen as I switched to
using the medium one I also got. See 3D Monster Maze!
I tried some drawing and partial update (and maybe full update) seem to
change a whole line at a time. See large screen 2nd pic.
You could use the image library to draw a white screen and then a vertical
black line at various positions to see if that clears it.
I also noticed that all the Pervasive demos show a full white screen then
black, then white. May need to cycle the screen with sleep in between.
I'll try tomorrow.
Have fun.
Nick

ste-r, 2016-02-26:
This is was a really useful thread, thank you to everyone. http://frederickvandenbosch.be was especially great.

PI Supply were telling reviewers last year that the finished products wouldn't have these issues. That indentation bug is a sore one to miss.

Pingk-NC, 2016-07-07:
I'm still having issues with this. So far I have...

Enable SPI interface through raspi-config

    sudo apt-get install python-imaging
    git clone https://github.com/PiSupply/PaPiRus.git
    cd PaPiRus
    sudo python**3** setup.py install 

    papirus-setup
    papirus-set 2.0

Running papirus-clock gives me:

    File "/usr/local/bin/papirus-clock", line 26, in <module>
    from papirus import Papirus
    File "/usr/local/lib/python3.4/dist-packages/papirus/__init__.py", line 1, in <module>
    from epd import EPD as Papirus
    ImportError: No module named 'epd'

However changing papirus-clock to

    - from papirus import Papirus
    + from papirus import Papirus as epd

Has no effect, and changing  __init__.py to

    - from epd import EPD as Papirus
    + from papirus import Papirus

Gives me the error:

    File "/usr/local/bin/papirus-clock", line 26, in <module>
    from papirus import Papirus# as EPD
    File "/usr/local/lib/python3.4/dist-packages/papirus/__init__.py", line 2, in <module>
    from papirus import Papirus
    ImportError: cannot import name 'Papirus'

So the "No module names epd" error has changes to "Cannot import name Papirus". 
I don't understand why this doesn't work, any help would be appreciated

shawaj, 2016-07-07:
When did you download the software?

Was it a while ago?

Thanks

Pingk-NC, 2016-07-07:
@shawaj 
All of this was downloaded/trialled yesterday/today, since my screen arrived yesterday.

shawaj, 2016-07-07:
Do you have a Pi3 to test with?

This is the first I've heard of it not working since we fixed it
previously.

When do the errors occur? Ideally need exact steps to reproduce the errors.

Thanks

shawaj, 2016-07-07:
Also please try with python 2 instead of python 3.

Thanks

Pingk-NC, 2016-07-08:
Unfortunately I don't have a Pi3 to test with.
The error occurs after installation, I have entered the following commands and all of them have gone through successfully:

    sudo apt-get install python-imaging
    git clone https://github.com/PiSupply/PaPiRus.git
    cd PaPiRus
    sudo python3 setup.py install 
    
    papirus-setup
    papirus-set 2.0

Then to test that the package worked, I ran

    sudo papirus-clock

and received the following error

    Traceback (most recent call):
        File "/usr/local/bin/papirus-clock", line 26, in <module>
            from papirus import Papirus
        File "/usr/local/lib/python3.4/dist-packages/papirus/__init__.py", line 1, in <module>
            from epd import EPD as Papirus
        ImportError: No module named 'epd'

Will try again with python2 and report back, however if you are working on python3 compatibility I would love to be a part of that effort.

zymurgybc, 2016-11-04:
I have an old display from the original crowd funding that was working until the last few days when I pulled a Raspbian update the bolluxed my poor 4gb SD card... I haven't found a way to not get the boatload of stuff.  Besides the point because I simply started with a bigger SD, pushed the new image... and none of my old code would run.  I started looking (aka googling) the problem with /dev/epd/ ... My fix was to stumble on the script `bin/papirus-setup` which I ran as `sudo bash bin/papirus-setup and all my errors stopped, and my old customized clock face popped up again. I got all my Python modules installed for both 2.7 and 3.4... but the missing device took persuasion :-)

francesco-vannini, 2017-01-18:
Could I please have an update of the issues. The thread is pretty old and things are likely to have been solved.
I'd like to close this in two weeks from now if there are no updates. Thanks.

francesco-vannini, 2017-02-14:
Closing the issue

closed 2017-02-14 by francesco-vannini