PiSupply/PaPiRus #4: papirus-write not working as expected

gocyclic249, 2016-01-24:
Writing to Papirus.......
Traceback (most recent call last):
  File "papirus-write", line 73, in <module>
    main()
  File "papirus-write", line 33, in main
    write_text(papirus, args.content, args.fsize)
  File "papirus-write", line 45, in write_text
    font = ImageFont.truetype('Pillow/Tests/fonts/FreeMono.ttf', size)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 240, in truetype
    return FreeTypeFont(font, size, index, encoding)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 137, in __init__
    self.font = core.getfont(font, size, index, encoding)
IOError: cannot open resource

gocyclic249, 2016-01-24:
I fixed this. The font variable in papirus-write was not pointing to the correct directory

shawaj, 2016-01-24:
Do you have the details of the correct directory and we will push it into
the next update?

Thanks

gocyclic249, 2016-01-24:
I added the freefont from debian repository an changed the following line in papirus-write

   font = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf', size)

shawaj, 2016-01-24:
What don't did you have to add and using what command? Just so we can
replicate the fix exactly.

Thanks

chrisrichards, 2016-01-24:
Just created PR #5 which fixes this, and a couple of other things

shawaj, 2016-04-01:
This should be fixed now :-)

closed 2016-04-01 by shawaj