PiSupply/PaPiRus #6: papirus-config not working as expected

hystrix1, 2016-01-31:
Using `sudo papirus-config` to set the screen size has no effect:

[image]

[image]

`sudo papirus-set` has the same issue:

[image]

hystrix1, 2016-01-31:
I just found a work-around this by editing papirus-set:
`sudo nano /usr/local/bin/papirus-set`, and editing 
`EPD_SIZE=2.7` to read `EPD_SIZE=2.0`

eichin, 2016-01-31:
By inspection, `papirus-set` can't actually work, because set_config as-written will only *change* an existing assignment, line 22 comments out any assignment that's already in the file.  I suspect "delete line 22 of papirus-set" fixes the whole problem, but I haven't tested it.

tvoverbeek, 2016-02-03:
The install (sudo python setup.py install) installs the wrong version of papirus-set.
THis is caused by the presence of the build directory.
Delete the build directory (under the main PaPiRus directory) and all the papirus-* files under /usr/local/bin/.
Then run the install again. This should install the right version (found in PaPiRus/bin.papirus-set).
The `EPD_SIZE=2.7` should now read `EPD_SIZE=$1` (line 25 in the right versuin).

francesco-vannini, 2017-01-18:
Closing this as it has been fixed.

shawaj, 2017-02-01:
A number of the functionalities of this still don't seem to work for examples these lines are obviously BS:

```
if [ "$OPTION" = "Web Interface" ]; then

	#!/bin/bash
	if (whiptail --title "Web Interface"  --yesno "Would you like to enable the web interface?" 10 60) then
	    echo "You chose Skittles Exit status was $?."
	else
	    echo "You chose M&M's. Exit status was $?."
	fi

fi

if [ "$OPTION" = "Test Mode" ]; then

	#!/bin/bash
	if (whiptail --title "Test mode"  --yesno "Do you want to enable auto test mode?" 10 60) then
	    echo "You chose Skittles Exit status was $?."
	else
	    echo "You chose M&M's. Exit status was $?."
	fi

fi
```

So will reopen this issue for now

shawaj, 2017-02-01:
That is from here - https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-config