PiSupply/PaPiRus #9: Generally not working

scargill, 2016-02-02:
Well, this was disappointing...

This repository does not seem to have been updated for months, the card that came with the display says "assembly guides, code examples, tips, tricks and much more" available at www.pi-supply.com - could someone tell me where because I could not even find an advert without searching - never mind the rest of it.

The font is in the wrong place, there is an indent fault in the WRITE page and so that won't work, the clock when it gets to 50 minutes after the hour starts to make mistakes on the time..

and more.

Check this - http://frederickvandenbosch.be/?p=1483

And even after fixing things thanks to that blog - I still can't get WRITE to work - as the command line reports an indent error after changing the font location - even though the indenting is correct.

Not at ALL happy.

YoungAtHome, 2016-02-02:
I've found indent errors on many print lines, perhaps as they were being
fixed for python3.
To check for indents, open in nano, move to second character and go down.
When the cursor jumps to the left on a full line then there's a tab to
replace. Check also in column 10.

Some fixes also documented in the papirus GitHub wiki pages.

I've got clock and gol working but still fixing font for write and calling
from python3.

Kevin also had these notes:
Here is my conclusion.

enable SPI by run raspi-config <<< mandatory!!!Rebootmodify
/usr/local/bin/papirus-clockfrom EPD import EPD

from papirus import Papirus as EPDmodify /usr/local/bin/papirus-writefont =
ImageFont.truetype('Pillow/Tests/fonts/FreeMono.ttf', size)font =
ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf', size)
add tab below two lines each head. text_lines.append("")
text_lines[current_line] += " " + wordNot sure to remove python3. I've
tested remove python3 and re-installed but the python3 was not a problem.

Nick

Well, this was disappointing...

This repository does not seem to have been updated for months, the card
that came with the display says "assembly guides, code examples, tips,
tricks and much more" available at www.pi-supply.com - could someone tell
me where because I could not even find an advert without searching - never
mind the rest of it.

The font is in the wrong place, there is an indent fault in the WRITE page
and so that won't work, the clock when it gets to 50 minutes after the hour
starts to make mistakes on the time..

and more.

Check this - http://frederickvandenbosch.be/?p=1483

And even after fixing things thanks to that blog - I still can't get WRITE
to work - as the command line reports an indent error after changing the
font location - even though the indenting is correct.

Not at ALL happy.

francesco-vannini, 2017-01-18:
We think all of the problems highlighted here have been fixed. Let us know if there are still issues with those.

closed 2017-01-18 by francesco-vannini