PiSupply/PaPiRus #10: papirus-write doesn't understand special characters

ungutknut, 2016-03-07:
When trying to output text containing ä ö ü (e.g. sudo papirus-write "ä") and most likely many more, wrong characters appear. Like when you choose a wrong charset.

Also I can't get a line break to work. \n doesn't seem to do the job.

theooos, 2016-05-20:
Same issue with the line break. Any solution found?

francesco-vannini, 2017-01-18:
The parsing of those escape character is not part of the current code implementation and needs refactoring of the code. I will investigate the issue regarding the lack of support for unicode.

shawaj, 2017-02-01:
The line break issue mentioned by @theooos is currently being discussed by @francesco-vannini @BenScarb @shawaj in #68 

Still need to sort other bits

cmseal, 2017-03-21:
Unicode support would be awesome, if possible...
Really love this project, and my Pi Zero W with PiPyRus is becoming so useful! :)

shawaj, 2017-03-21:
Glad you think so :-)

cyberscribe, 2017-04-22:
+1 for unicode or at least utf-8 support

shawaj, 2017-04-22:
@cyberscribe - at the moment it's not particularly high on our priorities list for PaPiRus as we're a little over stretched at the moment.

Would welcome some community assistance on this and some of the other issues if there is anyone out there that can help :-)

cyberscribe, 2017-04-22:
Is the driver a C/C++ module for Python, or is it written in pure Python?

cyberscribe, 2017-04-22:
Sorry, I see you're wrapping PIL.ImageFont, which does look like it can support unicode if done properly: http://stackoverflow.com/a/22612295

cyberscribe, 2017-04-22:
@shawaj if I were to augment PapirusText and PapirusTextPos to support utf-8 would you consider a pull request?

shawaj, 2017-04-22:
Oh 100% yes :-)

The ePaper driver is written in C and is over at the RePaper/gratis repo.

Everything here is python or bash

shawaj, 2017-04-22:
See #112

francesco-vannini, 2017-04-27:
This is now solved in #112

closed 2017-04-27 by francesco-vannini