PiSupply/PaPiRus #11: IndentationError

ced510xe, 2016-03-09:
Hello,
I am just starting with linux Raspberry etc...

I try to use a raspberry for a personnal project but i have this error :

root@raspberrypi:~# sudo papirus-clear
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clear", line 3, in <module>
    from papirus import Papirus
  File "/usr/local/lib/python2.7/dist-packages/papirus/__init__.py", line 2, in <module>
    from text import PapirusText
  File "/usr/local/lib/python2.7/dist-packages/papirus/text.py", line 37
    text_lines.append("")
             ^
IndentationError: expected an indented block

Please Help me!!

YoungAtHome, 2016-03-09:
There are tabs instead of spaces in that file.
To edit the file use:
sudo nano /usr/local/lib/python2.7/dist-packages/papirus/text.py
Replace with spaces all white space at the beginning of line 37 and also
the line before for good measure.
Best is also then to put the cursor on column 3 and move down line by line
looking for it jumping to the right. Replace whitespace.

I spoke to Aaron of Pi Supply at the Pi Party and we are looking to improve
GitHub and instructions. Keep going and just ask if you have any more
problems.

Good luck

Nick

therevoman, 2016-03-09:
Thank you for the information, but what about fixing the files in githib?

YoungAtHome, 2016-03-09:
Not got access yet but the Github wiki and issue #3 had lots of info.

shawaj, 2016-04-01:
This should be fixed now :-)

Could you re-download and confirm?

shawaj, 2016-06-14:
Closing this now as I do not think this is still an issue

closed 2016-06-14 by shawaj