PiSupply/PaPiRus #12: Switches not working

chrisrichards, 2016-03-09:
I can't get any of the switches to work, is there any sample code? According to  https://github.com/ukscone/PaPiRus-old/blob/master/notes.md they are on GPIO pins 25, 27, 28, and 29 but testing for continuity with a multimeter on these pins fails.

shawaj, 2016-03-15:
@chrisrichards - switches are on pins 36, 37, 38, 40 (GPIO 16, 26, 20, 21)

Hopefully this should fix your issue with them not working?

closed 2016-03-15 by shawaj

chrisrichards, 2016-03-16:
Yep they work! https://github.com/yellowfeather/PaPiRus/blob/master/bin/papirus-buttons.py

shawaj, 2016-03-16:
@chrisrichards - you should submit a pull request for that script and we can pull it in as an extra demo

theooos, 2016-06-02:
I'm having a strange bug with this script, and I assume it's down to the fact I'm using a Raspberry Pi 1B and so there are fewer GPIO pins. Two Three and Four are continually written to the screen, with no change in behaviour when any of the buttons are depressed. Do you have any ideas?