PiSupply/PaPiRus #14: IO error When "cold" start

ced510xe, 2016-03-11:
Hello

I have this error when I start my programm :
Traceback (most recent call last):
  File "/home/pi/Desktop/Test/Test.py", line 137, in <module>
    main(sys.argv[1:])
  File "/home/pi/Desktop/Test/Test.py", line 58, in main
    epd = EPD()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 66, in __init__
    with open(os.path.join(self._epd_path, 'version')) as f:
IOError: [Errno 2] No such file or directory: '/dev/epd/version'

With a "sudo papirus-setup" it work after but where are the base problèm??

YoungAtHome, 2016-03-12:
No dev EPD means device service is not running.
Check the logic in the papirus-config as the start /stop code logic may be
backwards, so you start it if it's running and stop it if not!

All the papirus Python code works by writing and reading through the device
files so /dev/EPD is crucial.

Let us know what you get working.

theooos, 2016-05-17:
Was having an identical problem to ced510xe above. We found that the solution for us was to add the following to the .bashrc:
`sudo service epd-fuse start`
Feel like this is a temporary fix, but saves running all of papirus-setup each time. Hope this helps :)

jflowers1974, 2016-05-20:
@theooos Thanks.  I was pulling my hair out - I seemingly could only get my screen to work after re-installing the driver each time.  At least now it works after a reboot.

johnhorner, 2016-12-11:
I'm trying `sudo service epd-fuse start` and all I get is 

`Job for epd-fuse.service failed. See 'systemctl status epd-fuse.service' and 'journalctl -xn' for details.`

When I do `systemctl status epd-fuse.service` I get this:

```
● epd-fuse.service - LSB: Start epd_fuse to control E-Ink panel
   Loaded: loaded (/etc/init.d/epd-fuse)
   Active: failed (Result: exit-code) since Sun 2016-12-11 16:30:44 AEDT; 34s ago
  Process: 2080 ExecStart=/etc/init.d/epd-fuse start (code=exited, status=1/FAILURE)
```
and when I do `journalctl -xn` all I get is `No journal files were found.`

**EDIT**: Fixed this by running papirus-config again and enabling EPD:
[Image](http://imgur.com/d5su11I)

And doing that plus adding the `sudo service epd-fuse start` to `.bashrc` seems to fix the issue even after restart.

shawaj, 2016-12-14:
What Pi version is this on? And latest firmware?

It shouldn't need anything changing to get it to auto appear on boot.

With a clean install I did today it survives every restart on stock noobs
on a Pi 3.

francesco-vannini, 2017-01-18:
I am going to close this issue as it is over a year old and no replies have been provided. I will assume the issue is resolved, please feel free to open again if not.

closed 2017-01-18 by francesco-vannini

MZandtheRaspberryPi, 2019-12-16:
FYI I was getting this error too, with a cronjob to start the papyrus e-ink display on reboot. Setting a sleep before starting the program for 20 seconds fixed it (you could probably do a shorter sleep).

This helped me figure out to let it sleep: https://www.raspberrypi.org/forums/viewtopic.php?t=216375