PiSupply/PaPiRus #15: Switches info and RTC Clock

ced510xe, 2016-03-11:
Hello 
I do'nt find any informations for the utilisation of the RTC or the manual switches near the screen.

Any body know somthing?

YoungAtHome, 2016-03-12:
RTC allows papirus to act as alarm clock with the pogo pin reseting or
starting the Pi (up to 2, not 3).

I haven't heard any Papirus users have got the switches working yet but
they could be used to change the state of the display or trigger update
pulls of information.

papirus-clock shows correct time so we can use that code to get the time

Some of us hope to get access to the  GitHub shortly.
Let us all know if you do anything with papirus.

ced510xe, 2016-03-13:
Hello

I just wand to change screen page when  one button is puched.

shawaj, 2016-03-15:
@ YoungAtHome - who wanted GitHub access?

Switches are on pins 36, 37, 38, 40 (GPIO 16, 26, 20, 21)

The RTC is a Microchip MCP7940N which you can see here - http://www.microchip.com/wwwproducts/en/MCP7940N

It is actually the same RTC as is used on the Pi Face boards. You can see more info about it here - https://github.com/piface/PiFace-Real-Time-Clock

This will allow it to be used with the standard Linux hwclock functionality.

Thanks

YoungAtHome, 2016-03-15:
Best would be myself, Frederick and the author if the papirus GitHub wiki
(if they want).

Thanks very much,
Nick

shawaj, 2016-03-15:
@YoungAtHome - is it you who I spoke to at the Pi Party about this? Can you link me to the PaPiRus GitHub wiki?

regilesknarf, 2016-03-16:
Hi,
made a short description on how to use the temperature sensor, RTC and the buttons on my blog:
[http://rpiland.blogspot.de/p/papirus-e-ink-display.html]

Kind regards
Frank

shawaj, 2016-03-16:
@regilesknarf - thanks very much for that, will take a look!

lxxl, 2017-01-26:
What about rasp zero and 2.0" panel and button's ? by using default script there is no all button's

Regards
lxxl

francesco-vannini, 2017-03-16:
This is all sorted now

closed 2017-03-16 by francesco-vannini