PiSupply/PaPiRus #18: text.py wrong indentation

francesco-vannini, 2016-03-21:
the indentation of the snippet below is still not right. Am I missing something here?

```
else:
                # No space left on line so move to next one
text_lines.append("")
                current_line += 1
text_lines[current_line] += " " + word
```

YoungAtHome, 2016-03-21:
Delete all characters to left of text including any (unexpanded) tabs and
hidden characters, and replace with the right amount of spaces.

Regards,
Nick

francesco-vannini, 2016-03-21:
Thanks Nick, I get that. What I meant is to understand why had the file not been fixed yet. Forgive me I am pretty new to GitHub but it seems quite a trivial fix.

shawaj, 2016-04-01:
This should be fixed now in the latest commit.

Can you try pulling down the latest code and testing it?

shawaj, 2016-04-11:
@francesco-vannini @YoungAtHome - can you verify this is now working ok?

shawaj, 2016-06-14:
Closing as this has been fixed.

closed 2016-06-14 by shawaj