PiSupply/PaPiRus #20: Text API example yields and error

francesco-vannini, 2016-04-02:
Running the following:
from papirus import PapirusText

text = PapirusText()

# Write text to the screen
# text.write(text)
text.write("hello world")

Will return:
Traceback (most recent call last):
  File "text.py", line 3, in <module>
    text = PapirusText()
TypeError: __init__() takes no arguments (1 given)

shawaj, 2016-04-02:
What about if you run papirus-write "hello world" from command line?

francesco-vannini, 2016-04-02:
The issue for text.py seems to revolve around inheritance. I have rewritten the file and made corrections.
import os
import sys

from PIL import Image
from PIL import ImageDraw
**from PIL import ImageFont**
from papirus import Papirus
import random

**WHITE = 1
BLACK = 0**

class PapirusText(Papirus):

    **def __init__(self):
        Papirus.__init__(self)**

    def write(self, text, size = 20):

        # initially set all white background
        **image = Image.new('1', self.size, WHITE)**

        # prepare for drawing
        draw = ImageDraw.Draw(image)

        font = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf', size)

        # Calculate the max number of char to fit on line
        **line_size = (self.width / (size*0.65))**
        
        current_line = 0
        text_lines = [""]

        # Compute each line
        for word in text.split():
            # If there is space on line add the word to it
            if (len(text_lines[current_line]) + len(word)) < line_size:
                text_lines[current_line] += " " + word
            else:
                # No space left on line so move to next one
                text_lines.append("")
                current_line += 1
                text_lines[current_line] += " " + word

        current_line = 0
        for l in text_lines:
            current_line += 1
            draw.text( (0, ((size*current_line)-size)) , l, font=font, fill=BLACK)

       ** self.display(image)
        self.update()**

shawaj, 2016-04-11:
have updated the repo for this now. should be working better? can you verify with a new clone?

ukscone, 2016-05-31:
the positioning of text is currently not working/implimented although you can mostly get an idea by looking in the demo directory in the repaper/gratis github entry

ukscone, 2016-05-31:
added a quick hack to get text positioning working in text.py. it's in my fork of your github

BenScarb, 2016-12-31:
I've put together some positional text stuff, in pull request https://github.com/PiSupply/PaPiRus/pull/68

francesco-vannini, 2017-01-18:
OK, I will close the issue as indeed the code was lacking the functionality and we are looking at implementing it with some refactoring. Thanks @BenScarb  for your efforts so far.

closed 2017-01-18 by francesco-vannini