PiSupply/PaPiRus #21: Wrong font path

francesco-vannini, 2016-04-02:
papirus-write under build/scripts-2.7/ still has the old font path

shawaj, 2016-04-02:
Think that file is not used at all?

francesco-vannini, 2016-04-02:
the files under build are used when setup is lunched. Removing the build folder before setup fixes the issue as a new build folder gets created with the correct files. I suspect that if a file is present it does not get overwritten.

shawaj, 2016-04-03:
So it seems like the best thing to do is remove the build folder completely then as we update software here, new installs will get the correct build files?

francesco-vannini, 2016-04-04:
Yes, it is working on my branch

shawaj, 2016-04-07:
Can you do a PR on that then? As the one you have done only includes an update to the inheritance problem, not the removing of the build folder

francesco-vannini, 2016-04-09:
I have created a PR but be careful as in the PR there is more than just the deletion of the build folder. There is also a review of the README and the text.py files.

shawaj, 2016-04-11:
whats wrong with readme and text.py?

shawaj, 2016-06-14:
@francesco-vannini is there still something wrong with readme and text.py?

If they are separate issues to this, could you put them in separate issues?

closed 2016-11-30 by francesco-vannini