PiSupply/PaPiRus #24: About framebuffer mirror function

ooxx-tw, 2016-04-15:
Hi,
I need framebuffer mirror function.
But, I execute papirus-config and select Mirror Framebuffer no action.
Please provide a tutorial or methods.
Thanks...

shawaj, 2016-04-16:
try typing `papirus-framepush` at the command line

ooxx-tw, 2016-04-17:
Command not found and I can not find papirus-framepush in PiSupply/PaPiRus/bin on GitHub.

shawaj, 2016-04-17:
Hmm that's odd! Will get it uploaded

lbendlin, 2016-04-30:
Any idea when you will be able to upload this?

ooxx-tw, 2016-04-30:
Mmm
Although the e-paper show slow response.
But I think the e-paper is to display clock and picture.
I need to synchronize LCD(Main) screen or became second screen.
And If the LCD screen sleep I can also see in the e-paper.

Other question:
I do not understand in the /dev/epd directory function.
This differs general LCD using the /dev/fbx.
Can you explain it?

/dev/epd/
├── BE
│   ├── current
│   ├── current_inverse
│   ├── display
│   └── display_inverse
├── command
├── current
├── current_inverse
├── display
├── display_inverse
├── LE
│   ├── current
│   ├── current_inverse
│   ├── display
│   └── display_inverse
├── panel
├── temperature
└── version

Thanks!

ukscone, 2016-05-31:
I made a few notes on the meanings of the directories in /dev/epd last year. http://blog.burninghorse.com/2015/07/24/a-few-notes-on-papirus/ that info was origiannly from Adafruit

ooxx-tw, 2016-11-09:
Hi shawaj,
Can you upload papirus-framepush?

shawaj, 2017-03-16:
Closing this issue as it is covered in #71 already 

Still on the to do list.

closed 2017-03-16 by shawaj