PiSupply/PaPiRus #25: use of gpio14 and gpio15 while PaPiRus is being used

0xygen8, 2016-04-17:
Just a quick question: will I be able to use the (hardware) serial port on pin 8 and 10 when the board is active? I have a project using the serial port and would like to add the PaPirus screen for output.
I see in the schematic that the pins are being used but can not determine if that is optional like the RTC

Thanks

shawaj, 2016-04-17:
The SPI is used for driving the display, however you should be able to use
more than one device on the SPI port.

shawaj, 2016-06-14:
Did you manage to get this working ok @0xygen8 ?

0xygen8, 2016-06-15:
Nope, the communication used on the serial port has a low baudrate and interferes with the display

shawaj, 2016-06-15:
SPI is definitely needed to drive the display so unfortunately it is not
optional. Sorry about that.

Can you get a similar device to your one but with an I2C or other interface?

Thanks

0xygen8, 2016-06-15:
Wow, we have taken the wrong exit here... I mean the TTL Serial port on the RPi (on pin 14/15) in the schematic the are connected to the board.

shawaj, 2016-06-15:
Pins 8 and 10 (GPIO 14 and 15) are used by the display yes.

The pin 8 labeled "border control" - it can probably work without this but
not without the discharge pin.

Thanks

shawaj, 2016-06-16:
Just to confirm - ideally both pins are needed. There is a 0.5mm width ring around the active area for mechanical tolerance connects to the Border. We reset this pin per update to keep it white, if you do not connect this, you will see this border ring getting gray after several updates.

The Discharge pin is to ensure display will be discharged completely which is required.

Closing this issue now as there is nothing we can do unfortunately.

closed 2016-06-16 by shawaj

0xygen8, 2016-06-16:
Clear Thank you for the update

shawaj, 2016-06-16:
No problem at all

aszolty, 2016-09-14:
Can the discharge and border control pins not be moved to use a different set of GPIOs on the pi? It'd be great to open up the TX RX pins for other serial operations if they're not needed specifically for that purpose.

shawaj, 2016-09-14:
No because they are fixed in the hardware.

In the future they could be moved but not right now.

Thanks

The1only, 2017-08-09:
This is a total NOGO, can I cut solder and move the p8 and p9 to tother IO pins and then revert the software??? 
Who came up with the idea of destroy the serial port just to get two IO pins, when any pins could be used???

shawaj, 2017-08-09:
@The1only that is the pins that the standard driver at https://github.com/repaper/gratis uses, so we used them too. Pretty simple really.

The UART is not really used that heavily. It is really only used by advanced users, so we don't see it as a particularly major issue...but we will probably fix it in the next hardware revision as there have been a couple of people asking about it.

You can change them to any other IO pins by cutting and resoldering yes - you will need to change the driver as well though.

Thanks

shawaj, 2017-08-09:
@The1only you could always post the issue at https://github.com/repaper/gratis also

tvoverbeek, 2017-08-09:
The https://github.com/repaper/gratis driver already has an option to free the uart. See https://github.com/repaper/gratis/blob/master/PlatformWithOS/RaspberryPi/epd_io_free_uart.h
There they use GPIOs 27 (pin 13) and 22 (pin 15). Note this is OK for Papirus Zero, but not for the HAT.
The HAT has the real time clock output connected to GPIO 27 (pin 13).

An other option to consider is to bit bang serial on any 2 available GPIOs.
This can be done e.g. using the pigpio library (http://abyz.co.uk/rpi/pigpio/).
See also https://raspberrypi.stackexchange.com/questions/27488/pigpio-library-example-for-bit-banging-a-uart with the answer from joan, the author of the pigpio library.
pigpio is included in Raspbian nowadays.

shawaj, 2017-08-09:
> PlatformWithOS/RaspberryPi/epd_io_free_uart.h
> 27488/pigpio-library-example-for-bit-banging-a-uart with the answer from

@tvoverbeek - what do you think with regards to changing the hardware?
Think it is worth doing for a future version of the HAT / Zero?

shawaj, 2017-08-09:
> tis/blob/master/PlatformWithOS/RaspberryPi/epd_io_free_uart.h

@tvoverbeek - and if the answer to my previous question is yes - what would
be the best rearranged pins and is it going to be relatively easy to
support in software having two different versions of the board with
different pins being used?

tvoverbeek, 2017-08-09:
Yes, I think it would be useful to free up the UART pins on the next version of the hardware (for eTC based displays).
The GPIOs are hard-coded in epd-fuse, but are easily changeable at compile time.
Will come back later with which alternative pins to use.
Maybe an option can be added to /etc/default/epd-fuse to select the pin arrangement for border control and discharge.
Otherwise we need 2 different versions of epd-fuse with the danger that the wrong version is used.

shawaj, 2017-08-09:
Sounds like a plan!

tvoverbeek, 2017-08-10:
Here my suggestion for rearranging the pins so we free the UART.
On both the Zero and the HAT GPIO 4 (pin 7) and GPIO 22 (pin 15) are routed to solder pads CN 11 and 12 respectively.
Why not routing GPIO 14 (UART0_TXD, pin 8) and GPIO 15 (UART0_RXD, pin 10) to CN11 and 12 and use GPIO 4 and 22 for border control and discharge?
This way the UART is available via the solder pads on both the Zero and HAT.
Moreover this solution would also work on the original 26 pin GPIO (minus the buttons)

shawaj, 2017-08-10:
Or if there's space we could just add 2 new pads so there's just some
extras?

tvoverbeek, 2017-08-10:
Yes, but you still need 2 different GPIOs for border control and discharge.
Still propose to use GPIO 4 and 22 for border control and discharge and not route them to solder pads any more.

shawaj, 2017-08-10:
And then just maybe route a couple others to pads if there's space?

Sounds like a good plan to me