PiSupply/PaPiRus #26: Please use '/usr/bin/python2' instead of '/usr/bin/python'

guydunigo, 2016-05-04:
For Archlinux, 'python' means 'python3', so please use '/usr/bin/python2' instead of '/usr/bin/python' in the papirus scripts.

shawaj, 2017-03-16:
@francesco-vannini @tvoverbeek do you know of a way whereby we could check for python version and server different code as a result? or would we just have to maintain to separate codebases?

shawaj, 2017-03-16:
See #52 also

guydunigo, 2017-03-16:
The problem is not about using python 3, I have both python2 and python3, it's just that /usr/bin/python is a link on linux but it can point to either /usr/bin/python3.x or /usr/bin/python2.x depending on the distro. (x being the installed version of python)
The thing is, there often is a link too named /usr/bin/python2 which always points towards /usr/bin/python2.x (available in raspbian too I think).
So for my issue, it is just about modifying #!/usr/bin/python to #!/usr/bin/python2.

shawaj, 2017-03-16:
Yeah sure, but they're similar enough that we can deal with them together
in one issue. Will leave this closed for now :-)

Thanks for reporting it though

shawaj, 2017-03-16:
Hey it wasn't even closed. We will leave it open for now then :-P

tvoverbeek, 2017-10-04:
See the python2-and-3 branch which has modified code which will run under both Python2 and Python3. Branch will soon be merged into master.

closed 2017-10-04 by tvoverbeek