PiSupply/PaPiRus #28: installation script error "error: file '/home/pi/PaPiRus/papirus-set' does not exist"

kurt77, 2016-05-08:
Am i the only one to receive this error:
error: file '/home/pi/PaPiRus/papirus-set' does not exist
after laubchibg the install script:
curl -sSL https://goo.gl/i1Imel | sudo bash
I've previously enabled the spi as requested.

shawaj, 2016-05-08:
Try again now. The path to papirus-set was wrong.

Thanks

shawaj, 2016-05-08:
Fix by changing path in setup.py to bin/papirus-set

closed 2016-05-08 by shawaj

kurt77, 2016-05-11:
Thank you...but now i have another error:
copying and adjusting bin/papirus-draw -> build/scripts-2.7
error: file '/home/pi/PaPiRus/bin/papirus-tmp' does not exist
bash: line 10: papirus-setup: command not found

kurt77, 2016-05-11:
there is another typo in papirus-setup:
bin/papirus-tmp
instead of
bin/papirus-temp
I could correct by myself in the git repo...but i'm not yet a git master...so i prefere you correct it.
Thanks