PiSupply/PaPiRus #30: papirus-temp: missing packages

kurt77, 2016-05-11:
papirus-temp needs "bc" package in order to work...maybe it should be installed during the install process.
It's also missing i2cget

pi@raspberrypi:~/Code $ sudo papirus-temp
/usr/local/bin/papirus-temp: line 8: i2cget: command not found
/usr/local/bin/papirus-temp: line 14: -5: substring expression < 0
(standard_in) 1: syntax error
(standard_in) 1: syntax error
Temp is  deg C ( deg F).

shawaj, 2016-05-12:
Can you make a pull request to fix these issues?

kurt77, 2016-05-12:
done!

shawaj, 2016-05-12:
Isn't i2c-tools installed by default in Raspbian? You just need to enable it from raspi-config I think?

kurt77, 2016-05-13:
It wasnt installed on my raspbian.
And yes you have to enable it with raspi-config

shawaj, 2016-06-14:
Fixed with #33

closed 2016-06-14 by shawaj