PiSupply/PaPiRus #37: Forum / Slack / Etc? Getting people connected

jflowers1974, 2016-05-20:
Don't know if this is well known or not - I just can't find it personally.  

But perhaps a link to a forum / slack / ??? for people to connect.   I'm trying to figure out how to use the buttons within a python script, and would love to know of a community (forum) I could hit up.

shawaj, 2016-05-20:
There is an example script for using the buttons called papirus-buttons
already on git.

We are in the process of getting a forum set up but it will take a couple
of months, at least, before it's ready.

shawaj, 2016-05-28:
Did you find the buttons example?

jflowers1974, 2016-05-29:
Hi there,

I did thank you.  I was actually more interested in the forum bit, and completely understand that this will take some time. 

With that said - I did have a weird issue.   I bought a new rPi 3, and as I was installing the hat - I noticed that there wasn't a post installed for GPIO 20 (or pin 38).  It just didn't come with one installed - and I was going to send the rPi 3 back, but decided to just make "SW3" another GPIO #.

However, I had zero success.  In the end I disabled SW3 by setting it to False.

Weird - right?...

shawaj, 2016-06-06:
Hmm - that sounds like a dodgy Pi 3...they should have all 40 pins of the GPIO populated.

The switch 3 is in hardware designed to be a certain GPIO so you can't change it to another one, other than by bypassing traces on the board with wire.

Keep an eye on our [Facebook](https://facebook.com/PiSupply) and [Twitter](https://twitter.com/PiSupply) for info on us starting a forum as we will definitely post it there :-)

closed 2016-06-06 by shawaj