PiSupply/PaPiRus #43: Add details about jumper configurations to documentation

shawaj, 2016-06-14:
Add details about jumper configurations to documentation for both PaPiRus HAT and PaPiRus Zero

francesco-vannini, 2017-03-16:
Done, both boards have extensive README.md files now

closed 2017-03-16 by francesco-vannini

projectarc-co-uk, 2017-06-05:
Any chance of some links to the relevant pages? Nothing is showing up for any jumper config info.

tvoverbeek, 2017-06-05:
For the PaPiRus HAT: https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20HAT/README.md
For the PaPiRus Zero: https://github.com/PiSupply/PaPiRus/blob/master/hardware/PaPiRus%20Zero/README.md
Where have you been looking?

projectarc-co-uk, 2017-06-05:
> For the PaPiRus HAT: https://github.com/PiSupply/
> PaPiRus/blob/master/hardware/PaPiRus%20HAT/README.md
> For the PaPiRus Zero: https://github.com/PiSupply/
> PaPiRus/blob/master/hardware/PaPiRus%20Zero/README.md
> Where have you been looking?

Hi,

Thanks for those links - that's what I was hoping to see.

I looked on the main doc page in the sections about setup and hardware
config (might be worth sprinkling some hyperlinks in all those pages to tie
the docs together a bit). I also looked on the Fred VDB pages about what he
had to do to get things working and also tried various Google searches
based around the terms "papirus jumper hardware config" - that issue log
was the only thing that turned up.

Sadly I'm still having troubles getting my PaPiRus Zero working - I was
hoping that it was just some extra jumper fiddling needed. I still get
ErrNo 103, after which the fuse service dies, causing an ErrNo 2 until I
restart the service, then it's back to ErrNo 103.

I'll keep thrashing about in the install scripts and stuff until I can get
something working, and add some extra comments then about whatever did the
trick. I guess it's all at early teething stages yet, I'm sure it'll be a
good little gadget once it works.

Thanks again for the links, at least I now know that's not the cause of my
pains!

Cheers.

shawaj, 2017-06-05:
It's not really teething problems, as it's fairly mature code and we rarely have complaints about it not working.

What OS are you using and how are you trying to install it?

projectarc-co-uk, 2017-06-05:
Oh right, that is is weird then.

I'm using a Pi Zero 1.3, with a standard NOOBS install of Rasbian. I did an
apt-get update & upgrade before running the auto-install script from the
main github page, did the raspi-config to set SPI and I2C, re-ran the
papirus-config script to make sure I had the right size (1.44), rebooted
and tried running the examples (mainly -clock and -write).

The first error I got was a ErrNo 103, after than I got ErrNo 2 until each
time I rebooted. After a few times round that loop of: install, configure,
ErrNo 103, ErrNo 2 and hunting round the web I realised that every time an
ErrNo 103 appeared, I'd always get an ErrNo 2 until the fuse service was
restarted or reinstalled.

So then I started searching some more for the ErrNo 103 issue. Only thing
that seems to get mentioned is that there's either some hardware or low
level config/driver step I'm missing. I've reseated the panel a few times
and tried moving the jumper, that didn't help. So I'm planning to start
hunting around in some logs to see if there's any hints hiding in any of
those, and after that I'll get hacking around in the driver stuff to see
what's going on.

At this point any help is appreciated, I'd assumed this was all new alpha
code, from all the comments around the web, hence my happiness to hack
about to get it working.

If it's all supposed to work out of the box, then I'm less impressed, cos
it doesn't work for me with a pretty standard setup :-(

Thanks for the reply, and thanks in advance if you have some suggestions
for what to try!

Cheers.

tvoverbeek, 2017-06-05:
Did you really run the auto-config script (https://github.com/PiSupply/PaPiRus/blob/master/install)?
Note, this script has to be run as root (using sudo).
The script already enables i2c and spi, runs papirus-setup which installs the driver and asks you for your screen size. So no need to run raspi-config and papirus-config manually.
Would suggest to start with a fresh Raspbian and run the auto-install as indicated in the README:
`curl -sSL https://pisupp.ly/papiruscode | sudo bash`
Then run the clock example.

projectarc-co-uk, 2017-06-05:
> PaPiRus/blob/master/install)?

Yup, definitely ran it - I can send you a full set of my history logs from
today if you really need proof. I ran the configs manually since various
pages mentioned needing to ensure I2C and SPI were enabled and the
dimensions were right, so I made doubly sure.

However since things didn't work according to the "auto" notes, I also
worked through the manual notes, and through all Fred VDB's notes, and the
gotchas page, hence me eventually getting to the point of wanting to check
the jumper was set right.

Given it looks like this should have just worked rather than being a fun
project, and it doesn't look like there's any logs or error messages
anywhere to give any hints as to the actual problem, I'm giving up with it
for now. When I next do a full reinstall I may pull it out and have another
play.

Sorry about that, I was hoping it'd be a nice little widget to play with :-(

projectarc-co-uk, 2017-06-05:
Hi,

Just to let you know I did a bit more digging into the EPD/Gratis and
general I2C side of things but didn't find any joy there.

However, I then remembered I had a Zero going spare from another project.
So I switched the PaPiRus and SD card into the spare Zero - and w00t! It
worked straight off :-)

So, my guess now is that I've either got a dodgy joint somewhere, or a
dodgy Zero. Time to resolder it and try again.

But anyway, my PaPiRus is happily running the demos now and I'm happy too.

So thanks for the replies and confirming I was doing the right things.

I don't know enough about I2C, SPI, or the display controllers - but if
there was some way to implement a low level setup test, that'd be really
helpful. To know whether it was the fuse layer, spi/i2c modules, faulty
physical connection, etc. would have prevented me going down a lot of wrong
paths.

Cheers.

shawaj, 2017-06-05:
@projectarc-co-uk glad to hear you are up and running!

That's also a great idea - maybe you can log this in a new issue as this one is already closed?

projectarc-co-uk, 2017-06-06:
Thanks Aaron, I've raised as issue #123 on the github project.

Cheers.

shawaj, 2017-06-07:
Thanks very much!