PiSupply/PaPiRus #57: Raspberry Pi 3

ChrisHaPunkt, 2016-09-06:
Hi all,
do not get this to work on my Raspberry Pi 3B model. 
Same steps on Raspi 2 and everything is working fine...

The install script running through as normal.
On console execution:
`pi@raspberrypi:~ $ sudo papirus-write "Test"`
`Writing to Papirus.......`
`Finished!`
But there is no update on the screen, just blank.
Someone get this to work on a Raspberry Pi 3?

auyongcheemeng, 2016-09-13:
I dont think the PaPiRus board works for the raspberry pi 3. the pin out for the pogo pin does not match for the Pi 3.

(For version 1.8) Not sure if the board you have have been revised

[board pics](http://imgur.com/a/uVVQd)

ChrisHaPunkt, 2016-09-13:
I'm running rev 1.9 where a support for the Pi 3 should be given (see pic at bottom, where u can see on the front left of the board the additional pogo pin for matching Pi3)

But I don't think the pogo pin is necessary for running the board. 
I removed the pogo pin for testing on the Pi 2 and it also works well.
Nevertheless i connected the pogo pin port on the Raspberry Pi 3, what didn't change a thing :(

[image]

shawaj, 2016-09-13:
Have you got SPI and I2C turned on?

Thanks

shawaj, 2016-09-13:
The v1.9 has Pogo pin locations for B+/2B/A+ and 3B so you can use with
either but this is not required to update the screen.

Thanks,
Aaron

ChrisHaPunkt, 2016-09-13:
Yes, both enabled:
[image]

shawaj, 2016-09-13:
Can you try command i2cdetect -y 1 ?

Does this find anything?

Thanks

ChrisHaPunkt, 2016-09-13:
I'l give it a try.

For comparison, the loaded modules on the working Pi 2:
[image]

ChrisHaPunkt, 2016-09-13:
I guess nothing has been found (Pi3):
[image]

ChrisHaPunkt, 2016-09-13:
As aspected, not even enabled.
[image]

But working fine:
[image]
[image]

shawaj, 2016-09-13:
Have you got another Pi 3 to try with?

Maybe the Pi 3 is the problem?

Does i2c command work on Pi 3?

Thanks

<https://cloud.githubusercontent.com/assets/4389395/18469128/31198d4c-79a7-11e6-8515-f35350bf2755.png>

<https://cloud.githubusercontent.com/assets/4389395/18469131/3724b3ec-79a7-11e6-855b-5145238723ad.png>
[image: img_20160913_114221]
<https://cloud.githubusercontent.com/assets/4389395/18469142/4520b86a-79a7-11e6-9f42-f17424d36ae4.jpg>

ChrisHaPunkt, 2016-09-13:
Not at hand, but i'll ask a friend of mine.
Can u confirm that it is running on another Pi 3?

shawaj, 2016-09-13:
Yes we have it running on many Pi 3 boards :-)

tedjackyx, 2016-09-17:
mine is 1.8
I just try on PI 3 and PI 2, both are not working.
I have no idea what happen on that...

shawaj, 2016-09-17:
Did it work before?

What error messages are you seeing?

Thanks

tedjackyx, 2016-09-18:
Hi, I didn't see any error,
i am useing:  
sudo papirus-write "123"
And got same response(commend) as you.

shawaj, 2016-09-18:
So what happens when you type that command and then hit enter? It works OK
or not?

ChrisHaPunkt, 2016-09-23:
May be some relations to this issue with latest jessie release on pi3:
https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=147291

I'll investigate further...

ChrisHaPunkt, 2016-10-06:
Since a simple loopback test failed on my Pi3, I tried the application on a new RPi3 and everything worked out-of-the-box ...
Going to send my RPi3 back .
Thanks all

closed 2016-10-06 by ChrisHaPunkt