PiSupply/PaPiRus #58: Writing text to a specific position not working

ChrisHaPunkt, 2016-09-14:
Taking the snippet from the docs:
`text.write("hello world", (10, 10) )`
results in

>   File "getEvents.py", line 26, in <module>
    text.write("hello world", (10, 10))
  File "/usr/local/lib/python2.7/dist-packages/papirus/text.py", line 24, in write
    font = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf', size)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 240, in truetype
    return FreeTypeFont(font, size, index, encoding)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 137, in __init__
    self.font = core.getfont(font, size, index, encoding)
TypeError: an integer is required

ChrisHaPunkt, 2016-10-12:
I'm not able to set the position of the text to write...
Even "                  " and tabs are not working and became trimmed.

BenScarb, 2016-12-04:
I'm getting the same thing.

The file:
/usr/local/lib/python2.7/dist-packages/papirus/text.py
doesn't have a function to take a (X,Y) parameter.

shawaj, 2016-12-05:
That particular function is broken. I think it only gives a font size at
the moment

dreadnoult, 2016-12-30:
It's pretty sad that such a basic functionality does not work out of the box and nobody tells you about that when you are buying the display. Nevertheless there is a simple workaround. You don't need the text module at all but you do need PIL (should be installed already if you have Papirus installed). Just do this:
'from papirus import Papirus 
from PIL import Image, ImageDraw, ImageFont
  
# The epaper screen object                                                                                            
screen = Papirus()                                                                                

# Create a new blank image with a size of  the screen and get a drawing context                                                                                   
image=Image.new('1',screen.size, 1)                                                            draw=ImageDraw.Draw(image)                                                                        
 
#  Specify a font you want to use and its size (last argument)                                                                                              
 font=ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeMono.ttf',32)              

# Draw text on the image at the desired position                                                                                                    
 draw.text((10,10),'Hello', font = font, fill = 0)                                             
 draw.text((10,60), 'World', font = font, fill = 0)                                           
 
#Update the screen                                                                                               
 screen.display(image)                                                                             
 screen.update() `

Hope it helps

BenScarb, 2016-12-31:
I've added in some positional text (PaPiRusTextPos) stuff, pull request is in.  See if it's the sort of thing you want.
https://github.com/PiSupply/PaPiRus/pull/68

francesco-vannini, 2017-03-16:
Implemented in #68

closed 2017-03-16 by francesco-vannini