PiSupply/PaPiRus #61: Cant install on raspberry pi 3 with latest raspbian

Palku, 2016-10-21:
I cant get my papirus to work.
I have used both installation methods mentioned but with no success.
The epd-fuse service wont start.

This is what papirus-setup outputs.
pi@raspberrypi:~ $ sudo papirus-setup
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libfuse-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
fonts-freefont-ttf is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'gratis'...
remote: Counting objects: 2548, done.
remote: Total 2548 (delta 0), reused 0 (delta 0), pack-reused 2548
Receiving objects: 100% (2548/2548), 8.31 MiB | 1.20 MiB/s, done.
Resolving deltas: 100% (1324/1324), done.
Checking connectivity... done.
Note: checking out 'ab46fd7'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at ab46fd7... [PlatformWithOS] simplify Clock27
make PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread  -lrt -o "gpio_test" gpio_test.o gpio.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd_test.o epd_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o spi.o spi.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd.o V231_G2/epd.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread  -lrt -o "epd_test" epd_test.o gpio.o spi.o epd.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd_fuse.o epd_fuse.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread  -lrt -o "epd_fuse" epd_fuse.o gpio.o spi.o epd.o
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C driver-common install
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
[ -e epd_fuse ] || false
install --group=root --mode=750 --owner=root epd_fuse "/usr/sbin"
[ -e /etc/default/epd-fuse ] || install --group=root --mode=644 --owner=root epd-fuse.default /etc/default/epd-fuse
if [ -e "../RaspberryPi/epd-fuse.conf" ] ; \
then \
  install --group=root --mode=644 --owner=root "../RaspberryPi/epd-fuse.conf" /etc/init ; \
fi
if [ -e "../RaspberryPi/epd-fuse.sh" ] ; \
then \
  install --group=root --mode=755 --owner=root "../RaspberryPi/epd-fuse.sh" /etc/init.d/epd-fuse ; \
  update-rc.d epd-fuse defaults ; \
fi
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
Job for epd-fuse.service failed. See 'systemctl status epd-fuse.service' and 'journalctl -xn' for details.

and this is what i get if i do "service epd-fuse status" 
● epd-fuse.service - LSB: Start epd_fuse to control E-Ink panel
   Loaded: loaded (/etc/init.d/epd-fuse)
   Active: failed (Result: exit-code) since Fri 2016-10-21 15:35:29 UTC; 3min 12s ago
  Process: 3920 ExecStart=/etc/init.d/epd-fuse start (code=exited, status=1/FAILURE)

Oct 21 15:35:29 raspberrypi epd-fuse[3920]: Starting EPD: epd_fusefuse: unkn...'
Oct 21 15:35:29 raspberrypi systemd[1]: epd-fuse.service: control process ex...1
Oct 21 15:35:29 raspberrypi systemd[1]: Failed to start LSB: Start epd_fuse ....
Oct 21 15:35:29 raspberrypi systemd[1]: Unit epd-fuse.service entered failed....
Hint: Some lines were ellipsized, use -l to show in full.

I dont know how to see the full error messages, my linux skills are about 0.

shawaj, 2016-10-25:
Is there a reason you are manually trying papirus-setup instead of the auto
install script?

Thanks

francesco-vannini, 2016-10-26:
Have you configured SPI and restarted?

mediumtech, 2016-12-13:
I have the same problem.

shawaj, 2016-12-13:
What have you done so far?

Thanks

Palku, 2016-12-14:
I downloaded a new image of "RASPBIAN JESSIE WITH PIXEL" and did the installation again.(Used the Lite first). That made it work for me.

mediumtech, 2016-12-14:
I've been using the latest Noobs from the Rpi website. Clean install or Raspbian from that, updated and upgraded. 

SPI enabled and restarted.

Tried the one-line setup and install first but get: 

bash: line 1: syntax error near unexpected token `<'
bash: line 1: `<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN" "http://www.w3.org/TR/html4/strict.dtd"><html lang="en"><head><meta http-equiv="content-type" content="text/html; charset=utf-8">'

So tried manually option 1, and get the same error as above.

Tried option 2, and get 2 errors. The file location cd /tmp/papirus/gratis-master/PlatformWithOS doesn't exist. I had a bit of a poke around and found the PlatformWithOS  folder, but the make then doesn't work.

I can try with Raspbian LITE, and/or the JESSIE image without Noobs if needs be, but would prefer to get it working the Noobs way as I want to take others through the project step by step who will also be coming at it with Pis set up using Noobs. I'd prefer not to have to start them off with a re-install unless we have to.

shawaj, 2016-12-14:
The one line setup should work fine on noobs.

Are you typing our copy pasting the command?

I just tried a clean noobs install and worked fine.

Need to enable i2c also

mediumtech, 2016-12-14:
JOY! 

I'd made a typo (doh), which was the cause of the first problem. I then made a clean install, enabled i2c and SPI, rebooted and tried again. Got an cert error, updated, then it ran but then just got the same error as @Palku . 

Tried again, this time updated, enabled i2c and spi, then rebooted before running the script. This time it worked, and I'm writing to the PaPiRus fine.

shawaj, 2016-12-14:
Fantastic! Glad to hear you're up and running again

francesco-vannini, 2017-01-18:
Have tested this with latest Pixel and Raspberry Pi3. It is all working for me. Would you guys be happy to close this issue?

closed 2017-01-19 by Palku