PiSupply/PaPiRus #62: Error with demos

dragongears, 2016-11-29:
I am trying to use the PaPiRus with a Raspberry Pi 2 and get the following error with any demo that writes to the screen. I am relatively certain I followed the setup instructions correctly.

pi@gear ~ $ sudo papirus-write "Test"
Writing to Papirus.......
Traceback (most recent call last):
  File "/usr/local/bin/papirus-write", line 73, in <module>
    main()
  File "/usr/local/bin/papirus-write", line 33, in main
    write_text(papirus, args.content, args.fsize)
  File "/usr/local/bin/papirus-write", line 69, in write_text
    papirus.display(image)
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 139, in display
    f.write(image.tobytes())
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 512, in __getattr__
    raise AttributeError(name)
AttributeError: tobytes

YoungAtHome, 2016-11-29:
tobytes() is a Python3 function so either use that, which is the best thing
to do, or find the future statements to add to your 2.7 code.

Nick

YoungInSurrey@gmail.com

francesco-vannini, 2016-11-29:
Hello, did you enable SPI?

-- 
Francesco Vannini

Website http://garagetech.tips
LinkedIn https://uk.linkedin.com/in/garagetechlinkedin
Youtube https://www.youtube.com/user/Happylotube
Twitter @GarageTechTweet
Facebook https://www.facebook.com/GarageTechBook/
<https://m.facebook.com/GarageTechBook/>
Google+ https://plus.google.com/+FrancescoVannini
Blog http:// <http://goog_1875614051/>blog.happylot.net

dragongears, 2016-11-30:
@francesco-vannini Yes, I enabled SPI.

dragongears, 2016-11-30:
@YoungAtHome I tried running the demo under Python3 and ended up getting even more errors.

It looks to me like the papirus-write demo is written in Python 2 but /usr/local/lib/python2.7/dist-packages/papirus/epd.py is Python 3. But I could be wrong since I don't know much about Python (yet). 

I haven't written any actual code yet. I got as far as setting up and installing the PaPiRus on a Raspberry Pi 2  running Raspbian and hit this roadblock while trying out the demos to make sure the PaPiRus works correctly.

francesco-vannini, 2016-11-30:
You could change the code to use tostring() instead of tobytes() or update your OS with sudo apt-get update or even dist-upgrade to get the latest firmware.

dragongears, 2016-12-01:
@francesco-vannini Changing the code to use tostring() did the trick! Thank you!

closed 2016-12-14 by francesco-vannini