PiSupply/PaPiRus #63: Specific instructions on how to attach the screen to the HAT?

johnhorner, 2016-12-10:
This is obviously too simple for most people, but I'm stuck. How do I connect the ribbon from the screen to the HAT? There's an obvious slot for it to go into, but it's just loose and comes out again. Is there some kind of clip involved? I'm concerned I'll break it.

Nowhere on the internet can I find instructions or a video on this first, most basic step. Anyone?

vogonsaudi, 2016-12-10:
I put some instructions in the wiki for this.  I didn't make a video, but hopefully this is helpful.
https://github.com/PiSupply/PaPiRus/wiki/Physical%20Assembly

johnhorner, 2016-12-10:
Thanks! But this is the part I'm not getting:

> The screen connects via a ribbon cable on the edge of the circuit board and bends around so the screen rests against the flat side.

*How* does it connect?

johnhorner, 2016-12-10:
I've got the answer now. The brown part of the connector on the HAT can be opened, gently lever it up, put the ribbon in then put it down again to clamp on the ribbon.

* in words: http://frederickvandenbosch.be/?p=1888
* video: https://www.youtube.com/watch?v=PiVEVRB3WxE&feature=youtu.be&t=131

vogonsaudi, 2016-12-10:
bingo!

shawaj, 2016-12-14:
Thanks for linking this.

Our new team member Francesco is working on full picture based setup guides
for all of our products, old and new, to make this process much easier for
everyone

francesco-vannini, 2017-01-18:
I'll close the issue as we will address the documentation in due course with a proper assembly guide. I will link the guide to the README.md as soon as it is made available.

closed 2017-01-18 by francesco-vannini

shawaj, 2017-02-01:
When they are available, they will be at https://pisupp.ly/papirus and https://pisupp.ly/papiruszero

Nothing there yet - but will be when ready