PiSupply/PaPiRus #64: It worked for a while, now I get epd errors. No idea what changed. Have reinstalled and rebooted.

johnhorner, 2016-12-11:
The display worked fine for a few hours after I set it up and installed using the automatic install.

Then for some unknown reason (nothing changed as far as I know) I can't do anything, and everything results in the error message below.

Just to repeat, everything worked, and I could use the clock and write and draw commands. Therefore the interface SPI was switched on and epd/fuse/whatever was present on the system.

Error message:

```
sudo papirus-write "foo"
Traceback (most recent call last):
  File "/usr/local/bin/papirus-write", line 73, in <module>
    main()
  File "/usr/local/bin/papirus-write", line 29, in main
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 66, in __init__
    with open(os.path.join(self._epd_path, 'version')) as f:
IOError: [Errno 2] No such file or directory: '/dev/epd/version'

```

johnhorner, 2016-12-11:
Needed to run `papirus-config` again for some reason to activate the EPD.

closed 2016-12-11 by johnhorner

shawaj, 2016-12-14:
You could also try sudo papirus-set 2.7 or whatever the screen size is next
time that happens too