PiSupply/PaPiRus #66: papirus-system and other examples

cmseal, 2016-12-14:
This is mentioned in the readme, but not in the bin folder. 

Likewise, any other examples from yourself or others would be really helpful, as I'm still learning python! :)

shawaj, 2016-12-14:
We will be adding more samples in the coming months to build the PaPiRus
platform :-)

cmseal, 2016-12-14:
Great, thanks for the quick response! Seemed like a cool piece of kit to play with, along with the servos and cameras I'm playing with using six Pi zeros! :)

Look forward to seeing some of the cool things that can be done with PaPiRus!

closed 2016-12-14 by cmseal

shawaj, 2016-12-14:
Thanks.

We should have a software developer on our team full time soon so our
output on that side will be hugely amplified :-)