PiSupply/PaPiRus #71: Add framebuffer mirror

shawaj, 2017-02-01:
Add framebuffer mirror example

shawaj, 2017-03-16:
Take a look at issue #24 also

francesco-vannini, 2017-07-25:
This is liked to https://github.com/repaper/gratis/issues/68 and https://github.com/raspberrypi/linux/pull/2119

shawaj, 2017-07-25:
> This is liked to repaper/gratis#68
> <https://github.com/repaper/gratis/issues/68> and raspberrypi/linux#2119

We should try this out tomorrow!

francesco-vannini, 2017-07-25:
I have tested it yesterday at the office but only managed to run the clock example from https://github.com/notro/pydrm More than this I could not really understand what to do really.

tvoverbeek, 2017-07-25:
Just did rpi-update and added the papirus overlay for the 2.7 screen.

The 2 demos (CounterDemo and clock27) work.
clock27 uses full update every minute (since it does drm.enable() and
drm.disable() around each update)
CounterDemo uses partial update since it does not do drm.disable() between
updates.

Note: the speed of CounterDemo is limited by the temperature dependent
stage-time
(same as in current repaper/gratis fuse implementation). Could reach higher
speeds
(as in the Pervasive video https://www.youtube.com/watch?v=enzUbiSWenQ) if
we could
set the stage-time directly independent of temperature.

The papirus module and the libfuse (/dev/epd) one cannot co-exist.
The kernel module also claims the lm75b interface, so any access from the
libfuse one fails.
This is the same issue as with the kernel rtc module loaded.

That is all for now

notro, 2017-08-01:
@tvoverbeek
I'm picking up from your comment in https://github.com/notro/tinydrm/issues/2 here so I don't clutter that announcement thread.

How are most people using these displays on the Raspberry Pi, as clocks?

tvoverbeek, 2017-08-01:
@notro 
Clock is only a demo app. Real applications are things like weather stations, twitter feed monitors, news and sports tickers, and anything else you can dream up.
Certainly a combination of text and (small) pictures/icons is a common application.
Hence my request to use dithering in the conversion to mono. It is a pity to throw away all the picture info you still have in the 8-bit grayscale after conversion from xRGB8888.
As far as I know dithering an already B/W image leaves the image unchanged.

Maybe Aaron, @shawaj and Francesco, @francesco-vannini can chime in with other application examples.

I am trying to demo this (adding F-S dithering to the repaper module) using your development instructions for RPi (https://github.com/notro/tinydrm/wiki/board:-Raspberry-Pi)

notro, 2017-08-01:
Here's a driver with dithering (before I decided against it) that didn't go anywhere: https://github.com/notro/tinydrm/blob/master/el320-240-36-hb-spi.c

You mentioned fbdev apps like fbcon and fbi, one thing to keep in mind is that the controller stays enabled  by default for 30 minutes before blanked, unless explicitly disabled.
Writing to /sys/class/graphics/fb1/blank between updates is one possibility.

tvoverbeek, 2017-08-01:
Thanks. Saves me some time. You did not use the 'real' Floyd-Steinberg, but the so called false one.
See http://www.efg2.com/Lab/Library/ImageProcessing/DHALF.TXT.
Will try with both variants and let you know the results.
Yes, I am aware of the console-blanking issue (have been bitten several times by it)

notro, 2017-08-01:
> You did not use the 'real' Floyd-Steinberg, but the so called false one.

I didn't know that, I just copied from an fbtft driver. I wasn't very impressed by the result, so that's probably the reason. But then again, I didn't know what to expect :-)

notro, 2017-08-03:
I have checked with the drm maintainer and we can't do software dithering in the kernel: https://lists.freedesktop.org/archives/dri-devel/2017-August/149243.html

tvoverbeek, 2017-08-03:
Noralf,
Tack för fråga på DRM listan / Thanks for asking on the DRM list.

I still think it is a pity for our use case, but I understand Daniel Vetter's reasoning.
Will try to find a user-space approach for e.g. displaying the pi-camera image as a dithered
gray-scale image on the Papirus displays.
I just got the real Floyd-Steinberg dithering (2 x 3 diffusion matrix) working in repaper.c yesterday.
Looks pretty good on the 2.7" display.

Anyway, thanks again for all your efforts / Tack så mycket för all dina ansträngningar.

Ton van Overbeek
PS Lived, studied and worked for 6 years in Sweden at Linköpings University

notro, 2017-08-04:
> PS Lived, studied and worked for 6 years in Sweden at Linköpings University

You picked up the language I see :-)

shawaj, 2018-04-07:
Think we can close this issue as it is fixed in https://github.com/PiSupply/PaPiRus/blob/master/bin/papirus-fbcopy ?

tvoverbeek, 2018-04-07:
Agreed

shawaj, 2018-04-09:
Fixed in #173

closed 2018-04-09 by shawaj