PiSupply/PaPiRus #78: I2C blocking

TobelRunner, 2017-03-10:
With the PaPirus hat installed I can no longer connect to other I2C devices. I am not sure if I misunderstood the solder pads for 3v3 5v and SDA and SCL but I thought these were for forwarding i2C and power to other devices. I was trying to connect a adafruit motor hat and PaPirus and I can only communicate with the PaPirus when both are connected. I then tried an BNO055 and again had no luck.

shawaj, 2017-03-10:
The pads for I2C do connect back to the Pi so there's no reason you
wouldn't be able to connect to other devices unless there was an I2C
address conflict.

francesco-vannini, 2017-03-15:
@TobelRunner - when you say "when both are connected" what exactly do you mean?

SDA, SCL, 5v0, 3v3, GND all have to be connected for it to work.

You may need to clean the pads though as they may have some resin residue on them (yellowy looking stuff)

closed 2017-03-15 by francesco-vannini

TobelRunner, 2017-03-15:
Hey Guys, Sorry for the late reply. In the end it was a wire shorting elsewhere. Adding the display to the project physically pushed a bunch of wires and due to my shoddy soldering some where shorted (and removing the display seemed to fix it). After a complete rewire everything is working fine, sorry for the issue and thanks for the comments.

shawaj, 2017-03-15:
No problem at all, glad you got it sorted :-)