PiSupply/PaPiRus #80: Text background

etatus, 2017-03-12:
Is it possible to write in-line white text with black background as it was selected (white text, black background)? It would be an handy feature.

shawaj, 2017-03-12:
So you just want a function where you can have say a sentence that is
partly white text on black background and partly black text on white
background?

etatus, 2017-03-12:
You don't need change parts of a sentence, just tell the write function if you want the normal text or inverted text.
The visual effect would be something like this:
[image]

shawaj, 2017-03-12:
Yes this seems like a good idea. I think what would be most useful is having two functions - one that does the whole text inverted and one that can do just a selection, similar to how bold would work in HTML or markdown.

Are you able to implement that?

etatus, 2017-03-12:
Frome here: http://stackoverflow.com/questions/18869365/a-watermark-inside-a-rectangle-which-filled-the-certain-color

I could get this piece of code:

```python
import Image
import ImageFont
import ImageDraw

img = Image.open("in.jpg")

draw = ImageDraw.Draw(img)
font = ImageFont.truetype(
    "/usr/share/fonts/truetype/ubuntu-font-family/Ubuntu-C.ttf", 66)
x, y = (width - 510, height-100)
# x, y = 10, 10
text = "copyright"
w, h = font.getsize(text)
draw.rectangle((x, y, x + w, y + h), fill='black')
draw.text((x, y), text, fill=(209, 239, 8), font=font)
img.save('out.jpg')
```
I think it would be easy to adapt to papirus, but I'm not familiar with papirus' code structure.

shawaj, 2017-04-03:
See #105 for potential fix

francesco-vannini, 2017-06-19:
This has been implemented in #105

closed 2017-06-19 by francesco-vannini

shawaj, 2017-06-19:
@francesco-vannini one thing that the readme doesn't make clear is whether you can do this for an entire screen only, or if you can do it only for certain parts of text?

Guessing it's just around that text a the rest of the screen is just left clear but maybe worth clarifying in readme? What do you think?