PiSupply/PaPiRus #81: Doesn't work on Raspberry Pi Zero W

fangfufu, 2017-03-13:
I have a PaPiRus Zero (PZ). I recently bought a new Raspberry Pi Zero W (RPi 0W). The commands used to control PZ successfully execute on RPi 0W. However, PZ doesn't react to it. I plug my PZ on my old RPi 0 along with the same microSD card, PZ works fine. So I am certain that there is some sort of compatibility issue.

shawaj, 2017-03-13:
@fangfufu have you installed the software on the Pi Zero W and turned I2C and SPI on?

Because we have tested it with the Pi Zero W and it works perfectly

shawaj, 2017-03-13:
@fangfufu - picture of it working with Pi Zero Wireless has - http://m.imgur.com/eYbnkQb

(Picture courtesy of @francesco-vannini )

shawaj, 2017-03-13:
*here

closed 2017-03-13 by shawaj

fangfufu, 2017-03-13:
@shawaj , I have installed the software in Pi Zero W and turned the I2C and SPI on. The same microSD card works fine on my old Pi Zero.

shawaj, 2017-03-13:
Please try with a fresh installation of Raspbian on the Pi Zero W as it's
working fine for us.

Thanks

<https://github.com/PiSupply/PaPiRus/issues/81#issuecomment-286253531>, or mute
the thread

fangfufu, 2017-03-13:
@shawaj I have tried flash a new microSD card with Raspbian Lite, and re-installed everything. It still doesn't work.

shawaj, 2017-03-13:
Try Raspbian full version instead of Raspbian lite :-)

fangfufu, 2017-03-14:
I don't understand why Raspbian full would make a difference, compared
to Raspbian lite.

Before I flash my microSD card again, is there a better way to diagnose
this problem? Where can I get the error messages?

shawaj, 2017-03-14:
Raspbian full and lite have many different packages installed by default,
among other notable differences.

It's best to try a like for like swap so that we can be 100% sure.

Thanks

fangfufu, 2017-03-14:
I have just flashed Raspbian full, and reinstall the drivers. It still
doesn't work... Once again, the same MicroSD card work for my Raspberry
Pi Zero... ``cat /dev/epd/temperature`` works though.

shawaj, 2017-03-14:
Seems like it should be working then if it's appearing.

I can't see an issue.

fangfufu, 2017-03-14:
If I understand it correctly, the temperature sensor uses I2C bus,
whereas the display itself uses the SPI bus... Anyway, it is not
working. Please tell me how to diagnose this. There is definitely an
issue...

fangfufu, 2017-03-14:
Could you send me your microSD card image? You could Dropbox / Google Drive it.

fangfufu, 2017-03-14:
I have two Raspberry Pi Zero W lying around. I have tried them both, none of them work. It works fine with the original Rasbperry Zero though.

shawaj, 2017-03-14:
As said, start from a fresh Raspbian full installation and follow the
instructions to install everything including turning in spi and i2c.

If it works on one, it works on all. They are pin compatible.

Please make sure also that you have a good connection. Some headers for the
Pi Zero are not very well made for example - make sure it's one who square
pins not flat pins.

Thanks

fangfufu, 2017-03-14:
> As said, start from a fresh Raspbian full installation and follow the
> instructions to install everything including turning in spi and i2c.
>
I have done exactly that.
> Please make sure also that you have a good connection. Some headers
> for the
> Pi Zero are not very well made for example - make sure it's one who square
> pins not flat pins.
I have soldered the header for all the Raspberry Pi Zeros that had been
tested.

Anyway, so far I have tested on 2 Raspberry Pi Zero Ws,and 1 Raspberry
Pi Zero. It works on my Raspberry Pi Zero, but not Zero Ws.

shawaj, 2017-03-14:
Seems like a loose connection is possible. Can you send a picture of the
header you have soldered in?

Thanks

fangfufu, 2017-03-14:
I really don't think there is any soldering problem. Do you not have any ideas on how to diagnose epd-fuse daemon? 

[image]

shawaj, 2017-03-14:
Unless you installed it wrong, EPD fuse should be working fine. It's no
different on pi zero to pi zero w.

Please send a picture of the header itself, not just the underside.

Thanks

fangfufu, 2017-03-14:
I haven't installed it wrong, because it is on the same microSD card. It is a fresh installation too. I have put my RPi ZW back into the enclosure. 
[image]

shawaj, 2017-03-14:
Are you sure it's not the enclosure clashing?

You should also use a *different* SD card for the Pi Zero to the Pi Zero W
for testing purposes just to rule out any issues there.

It seems to either be a connection issue or a software installation issue.

Thanks

fangfufu, 2017-03-14:
I have not been using that enclosure when I am using the Papirus Zero. I
have put my RPi ZW back in that enclosure, because I don't think I am
going to get Papirus working with RPi ZW.

I don't see how using a different SD card for Pi Zero and Pi Zero W
would rule out any issue, other than adding in more variable. I ran your
script under under RPi ZW, it didn't work. Then I put the card in RPi Z,
it worked. I have also tested my Papirus Zero on two separate RPi ZW,
the both didn't work consistently . So I think it is clear that there is
some sort of compatibility issue.

Why don't you send me your SD card image to rule out software problem?

shawaj, 2017-03-14:
Because we've already tested it and it works flawlessly :-)

fangfufu, 2017-03-14:
And before you blame "enclosure clash", the first image actually has Papirus attached to my RPi ZW, with the Papirus at the bottom... 

Anyway, I give up.

shawaj, 2017-03-14:
We will try again tomorrow with some more Pi Zero Ws if we can get our
hands on a couple more.

See if​ we can reproduce anything.

shawaj, 2017-03-14:
We have tested with every Raspberry Pi Zero Wireless we have in the office and it works with all of them using the instructions provided here.

So it's not an issue with our software or with Raspbian.

It must either be a loose connection, that you have not installed the software correctly or that you have broken some of the GPIO pins on your Pi Zero Ws from soldering them too hot or something.

shawaj, 2017-03-14:
@francesco-vannini also tested the board by installing the software on a pi zero normal, and then transferring the PaPiRus Zero and the SD card to the Pi Zero Wireless and that worked also. So again it must be an issue with one of the things mentioned above

fangfufu, 2017-03-14:
Well, I performed SPI loopback documented here: https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md

It worked fine.

shawaj, 2017-03-14:
> Well, I performed SPI loopback documented here:
> https://www.raspberrypi.org/documentation/hardware/
> raspberrypi/spi/README.md
>
> It worked fine.

Glad to hear you got it working. Great news :-)

fangfufu, 2017-03-14:
No, it still doesn't work. : ( I can only confirmed that the SPI bus
works, and the epd-fuse freeze up my serial console.

fangfufu, 2017-03-14:
I am getting my 3rd Raspberry Pi Zero W... If it works, I will let you guys know...

shawaj, 2017-03-14:
Are you sure you are using a clean Raspbian install? With nothing else
installed?

And you're using the one line install from our main GitHub readme?

fangfufu, 2017-03-14:
Yes, I am 100% sure. I am getting a new Raspberry Pi Zero W, because the
one I have right now refuse to enter gadget mode. So the supplier
allowed me to return it.

shawaj, 2017-03-14:
Hmm strange. Maybe you just had really bad luck and it was a faulty one :-)

fangfufu, 2017-03-14:
Surely I can't be unlucky enough to get two faulty Raspberry Pi Zero W
in a row, right? :(

shawaj, 2017-03-14:
Dunno but we've got the software working on pi zero wireless and you said
the board is working on pi zero.

So points to a software configuration or pi zero issue on your end. I can't
see what else it could be

shawaj, 2017-03-14:
@fangfufu just had a thought - are you trying to connect to this through serial console and use the Pi Zero Wireless at the same time? If so I don't think that will work.

Try it with a screen, keyboard, mouse etc (or over SSH) first and then let us know if that sorts your issue.

fangfufu, 2017-03-14:
I have been trying to use the Raspberry Pi Zero gadget mode. Does that
change things?

fangfufu, 2017-03-14:
I have been using gadget mode with Raspberry Pi Zero. Papirus works fine.

shawaj, 2017-03-14:
Perhaps you could try with Pi Zero W not in gadget mode (using it just normally with a screen and KB/mouse) and report back

shawaj, 2017-03-14:
then we can try to isolate the issue

fangfufu, 2017-03-15:
It still doesn't work with Pi Zero W not in gadget mode. Is there any sort of logs produced by the driver?

shawaj, 2017-03-15:
Must be a loose connection then or broken GPIO pins. Can't really see how
it could be anything else

fangfufu, 2017-03-15:
So which pin is broken then? Can the driver tell?

shawaj, 2017-03-15:
No sorry. There's not any more we can do to help. You will have to work
that out yourself.

But we are 100% sure it works on pi zero wireless as we have them running
here and the PaPiRus itself works as you said.

As said, start from a FRESH Raspbian / NOOBS installation and follow the
install one line instructions on the main readme of this GitHub. Do not
install any other software or do anything other than enable I2C and SPI and
it will work assuming that your Pi Zero is not faulty or the GPIO pins have
not been damaged.

Thanks

fangfufu, 2017-03-15:
Hey, I got my 3rd Pi Zero W. Now everything is working. It turns out that my 2nd Pi Zero doesn't have 5.0 on one of the pin used by Papirus. This whole experience has been stressful. I can't believe that my two Pi Zero Ws from two different suppliers are slightly broken in different ways...

shawaj, 2017-03-15:
Perfect, glad you have it working now

amkeener, 2018-09-24:
Just a heads up... I experienced this same issue on two Raspberry Pi Zero Ws... The only solution that worked for me was to disable both serial options in raspi-config (serial login shell and serial port hardware).