PiSupply/PaPiRus #82: Latest Kernel (4.9.x) on Raspberry Pi Zero W

Dom1n1c, 2017-03-15:
I noticed that the latest kernel avoids communication on the pi zero w.

to revert to 4.4.50:
rpi-update 52241088c1da59a359110d39c1875cda56496764

shawaj, 2017-03-15:
What do you mean it "avoids communication on the Pi Zero W"?

Dom1n1c, 2017-03-15:
```
root@ePaper:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    288      0 --:--:-- --:--:-- --:--:--   289
100 53.6M  100 53.6M    0     0   682k      0  0:01:20  0:01:20 --:--:--  992k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.14-v7+
 *** depmod 4.9.14+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to b30461d2621e71eb9b8845b38738af177b9181ed
 *** A reboot is needed to activate the new firmware

 
root@ePaper:~# papirus-clock
panel = EPD 2.0 200 x 96  version=4 COG=2 FILM=231
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 123, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 70, in main
    papirus.clear()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 152, in clear
    self._command('C')
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 156, in _command
    f.write(c)
IOError: [Errno 103] Software caused connection abort

root@ePaper:~# papirus-clock
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 123, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 66, in main
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 66, in __init__
    with open(os.path.join(self._epd_path, 'version')) as f:
IOError: [Errno 107] Transport endpoint is not connected: '/dev/epd/version'
```

francesco-vannini, 2017-03-15:
Try reinstalling papirus after the update.

Have you also got SPI and I2C enabled?

Also try `fusermount -u /dev/epd`

shawaj, 2017-03-15:
Hmm yes I guess that's true but you should submit an issue on the Raspberry
Pi GitHub if you have found an incompatibility issue with a driver :-)

pelwell, 2017-03-16:
Have you tried with the most recent firmware release (b30461d2621e71eb9b8845b38738af177b9181ed - Mar 13)? It fixes a clocking issue that caused I2C clocks to often be 60% faster than requested, which is enough to break some applications.

shawaj, 2017-03-17:
Reopening this issue as it seems like it will need further investigation

shawaj, 2017-03-17:
@francesco-vannini have you got a Pi Zero with PaPiRus already setup where you could update the firmware to the latest version and check operation?

Dom1n1c, 2017-03-17:
@pelwell yes I tried the most recent firmware https://github.com/Hexxeh/rpi-firmware/commit/b30461d2621e71eb9b8845b38738af177b9181ed
but it works only up to https://github.com/Hexxeh/rpi-firmware/commit/52241088c1da59a359110d39c1875cda56496764

shawaj, 2017-03-17:
For reference, also see this issue over at https://github.com/raspberrypi/linux/

https://github.com/raspberrypi/linux/issues/1902

shawaj, 2017-03-17:
@Dom1n1c - is this **only** on the Raspberry Pi Zero?

Dom1n1c, 2017-03-17:
@shawaj i only have a few pi  zeros / and pi zero w to use with the PaPiRus Zero
(Of course it would fit on a bigger Pi)

shawaj, 2017-03-17:
@Dom1n1c ahh ok - so you don't have a Pi 3 to test with?

Thanks

Dom1n1c, 2017-03-17:
I could temporarily swap it and see if it works.
With the latest raspbian realease without a succeeded rpi-update everything works fine.

shawaj, 2017-03-17:
@Dom1n1c - tried it with a PaPiRus HAT on Pi 3 with b30461d2621e71eb9b8845b38738af177b9181ed and that also seems to fail

shawaj, 2017-03-17:
@Dom1n1c - have found the issue in https://github.com/raspberrypi/linux/issues/1902 with the kind help of @pelwell and @tvoverbeek 

Will push the updates to the https://github.com/repaper/gratis repo later

Or have you already done this @tvoverbeek ? If not I can handle that now

shawaj, 2017-03-17:
Fixed here - https://github.com/repaper/gratis/pull/64

Thanks all!

closed 2017-03-17 by shawaj