PiSupply/PaPiRus #84: Portrait mode

daxx2k, 2017-03-16:
Hi there, I would like to use the large 2.7inch display in portrait mode, there is any way I can display text in that way? thanks!

shawaj, 2017-03-16:
This is an interesting request that we've not had before. Perhaps @BenScarb and @francesco-vannini might have some ideas about how to do this?

In the meantime you could perhaps turn the text into an image and have it rotated?

shawaj, 2017-03-17:
Closing this issue as issue #90 has been opened but it's more generic and think more relevant to have it so that you can rotate the screen.

In framebuffer / fbtft this can be done by specifying clockwise rotations from standard - 90, 180, 270 - think this would be our best bet

closed 2017-03-17 by shawaj

daxx2k, 2017-03-27:
thanks for looking into that!

shawaj, 2017-03-27:
@daxx2k - if you go to that other issue linked above and subscribe to it, then you will get an update once we build in the support :-)