PiSupply/PaPiRus #85: Perhaps refactor papirus-temp with device tree driver for LM75

shawaj, 2017-03-16:
Perhaps we could look at using the newly created LM75 temp sensor device tree driver and update the papirus-temp script accordingly?

In future, this could be incorporated as a device tree blob onto the EEPROM on the PaPiRus HAT

shawaj, 2017-03-16:
Take a look here - https://www.raspberrypi.org/forums/viewtopic.php?p=1130515#p1130515

And here:

https://github.com/raspberrypi/linux/search?utf8=✓&q=Lm75

shawaj, 2017-03-16:
Also see #41 

Would be good to change this to a python script rather than a bash script (could keep the bash script as a legacy script somewhere, as seems pointless to just flat out delete it but worth having more continuity in programming language used)

francesco-vannini, 2017-05-17:
@tvoverbeek is this taken care of in #117 ?

tvoverbeek, 2017-05-17:
No. The I2C interface for the LM75 (as long as you are only interested in the temperature) is very simple. Does not warrant to use the device-tree for this simple interface.
Converting it to a python script is something worth doing.

shawaj, 2017-05-17:
The device tree driver is already there I believe? Just need to add a line to the config.txt I think to enable it

tvoverbeek, 2017-05-17:
Yes, but when enabled you cannot access it directly via I2C in python using smbus.. Would have to rewrite the LM75B access in lm75.py (imported by epd.py) to make use of the kernel driver interface. Moreover installation then requires checking of config.txt for the correct dtoverlay.

shawaj, 2017-05-18:
@tvoverbeek cool - lets park this one for now then. Seems like a bad idea!