PiSupply/PaPiRus #90: Invert display so the Pi can have any orientation

mashedkeyboard, 2017-03-17:
I'd love to be able to use PaPiRus with my Pi Zero "upside down" as considered by the PaPiRus HAT - with the text being displayed the right way up.

Could we have a "setOrientation()" in a future build?

shawaj, 2017-03-17:
Related to #84

shawaj, 2017-03-17:
In framebuffer / fbtft this can be done by specifying clockwise rotations from standard - 90, 180, 270 - think this would be our best bet

Dom1n1c, 2017-03-20:
i would like to have an option which allows me to rotate the screen by 180° 
(to attach the RPi Zero W to the Wall and to have the cable running underneath)

shawaj, 2017-03-20:
Yes this is a very good idea for sure. We will definitely look into it.

Dom1n1c, 2017-03-24:
Hey @shawaj 
Now, since the PaPiRus is working with the latest kernel on my Pi Zero W, I would like to have the correct orientation on my ePaper. 
Is there any way to do this before: image.write? 
I only display Text right now, so inverting the font would be to complicated. 
Instead: Displaying an image could easily be manipulated.

shawaj, 2017-03-24:
Do you need to write text on the fly?

Otherwise​ you could generate images, save then upside down, and send to
screen?

Dom1n1c, 2017-03-25:
Thats my code:
https://github.com/Dom1n1c/PaPiRus-MQTT/

shawaj, 2017-03-25:
What does the code do? If you want it included in the PaPiRus repository
can you submit a PR

Thanks

Dom1n1c, 2017-03-25:
it listens for mqtt topics and then it'll print them to the display 😄 
Here you can find a picture of the PaPiRus Zero running this script
https://community.openhab.org/t/using-an-epaper-eink-display-as-energy-temperature-monitor-mqtt/16954/2

Dom1n1c, 2017-05-06:
Any news on rotating the screen (changing the orientatiion)?

Thanks :)

francesco-vannini, 2017-05-07:
Hi @Dom1n1c, I believe that would in fact require some work on the EPD driver what do you think @tvoverbeek ?

tvoverbeek, 2017-05-07:
@Dom1n1c @francesco-vannini  No need to modify the repaper/gratis driver.
Should be possible to handle this on the python side using the PIL library.
But requires careful splitting of physical screen coordinates  and size (post rotation) and virtual screen coordinates and size (pre rotation).
180 deg should be easier (no change in screen height and width), 90 and 270 requires more work

quavoce, 2017-05-14:
I too would really like this feature please. 
My use case would be to have dynamically updating text aligned next to the buttons on a PiZero, in portrait mode with the 2″ diagonal (200 x 96 resolution) display. 
If I could write text in portrait mode (90 degrees), I would position the text using x/y values.
I do realise I could use multiple images for each button status, but having dynamic text would be easier.
Thanks!

tvoverbeek, 2017-05-14:
See my PR #119 Hopefully will get merged soon. Otherwise clone my repository

quavoce, 2017-05-14:
Brilliant! Can't wait to try this out, thank you!

shawaj, 2017-05-14:
Think @francesco-vannini was playing with this but got distracted with some other stuff at tail end of last week. Will get it merged this coming week :-)

francesco-vannini, 2017-05-14:
We are reviewing a PR for this and we are aiming to merge next week.

francesco-vannini, 2017-05-17:
Sorted in #119

closed 2017-05-17 by francesco-vannini