PiSupply/PaPiRus #91: PaPiRus textpos does only a partial update?

shawaj, 2017-03-18:
PaPiRus textpos does only a partial update, so if there's something over the screen.

@BenScarb @francesco-vannini do you think it would be better to have it with a full update, or have an option to do either a full or partial update? I like the idea of a partial update as it allows you to write text over an image but sometimes you just want text...

What do you think?

francesco-vannini, 2017-03-21:
I think textpos.py should default to full update and have an option for partial
Changed textpos.py to only do full updates in #94 until the option is introduced.

shawaj, 2017-03-21:
Yes, agreed @francesco-vannini - the partial should be an argument. This should probably be the case with all example scripts

francesco-vannini, 2017-05-17:
@BenScarb this has been dealt with right?

francesco-vannini, 2017-06-19:
I think this could be closed

shawaj, 2017-06-19:
Did we fix it already?

shawaj, 2017-06-19:
@benscarb was this fix part of #105?

BenScarb, 2017-07-23:
I do believe it was, commits on the 4th of Apr allow for partial or full updates to be done.  Defaults to Full update.
https://github.com/PiSupply/PaPiRus/pull/105#commits-pushed-609ddbc

francesco-vannini, 2017-07-25:
Thanks @BenScarb, closing this up.

closed 2017-07-25 by francesco-vannini