PiSupply/PaPiRus #99: Papirusimage.write only takes 2 args

saty9, 2017-03-23:
the readme docs describe a 3 argument version of papirusimage.write
image.write('/path/to/image', 20, (10, 10) )
however that doesn't seem to exist.

shawaj, 2017-03-23:
@francesco-vannini can you shed any light on this?

Is there a 3 argument version?

GregLukosek, 2017-04-03:
This is very disappointing. Why does doc describing how to use methods that aren't existing?

GregLukosek, 2017-04-03:
to clarify. PapirusImage() does not contain any method except simple write that draws image full screen

shawaj, 2017-04-03:
This is in a PR I believe which we will be looking at this week.

tvoverbeek, 2017-10-04:
See commit 48ca0fc on the papirus2-and-3 branch.
image.py and README updated.
Something similar to the not-implemented method in PapirusImage is available in PapirusComposite.
Closing this issue.

closed 2017-10-04 by tvoverbeek