PiSupply/PaPiRus #100: papirus-setup fails (epd fuse)

Dom1n1c, 2017-03-25:
```
papirus-setup
Reading package lists... Done
Building dependency tree
Reading state information... Done
libfuse-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
fonts-freefont-ttf is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'gratis'...
remote: Counting objects: 2569, done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 2569 (delta 4), reused 0 (delta 0), pack-reused 2555
Receiving objects: 100% (2569/2569), 8.32 MiB | 938.00 KiB/s, done.
Resolving deltas: 100% (1337/1337), done.
Checking connectivity... done.
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o gpio_test.o gpio_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o gpio.o ../RaspberryPi/gpio.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread  -lrt -o "gpio_test" gpio_test.o gpio.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd_test.o epd_test.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o spi.o spi.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd.o V231_G2/epd.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread  -lrt -o "epd_test" epd_test.o gpio.o spi.o epd.o
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"'   -c -o epd_fuse.o epd_fuse.c
cc -D_FILE_OFFSET_BITS=64 -I/usr/include/fuse  -Wall -Werror -std=gnu99 -I../RaspberryPi -IV231_G2 -I. -DEPD_IO='"epd_io.h"' -lfuse -pthread  -lrt -o "epd_fuse" epd_fuse.o gpio.o spi.o epd.o
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
make DESTDIR="" PREFIX=/usr SERVICE=systemd PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C PlatformWithOS/driver-common install
make[1]: Entering directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
[ -e epd_fuse ] || false
[ ! -d "/usr/sbin" ] && mkdir -p "/usr/sbin" || true
install --group=root --mode=750 --owner=root epd_fuse "/usr/sbin"
[ ! -d "/etc/default" ] && mkdir -p "/etc/default" || true
[ -e "/etc/default/epd-fuse" ] || install --group=root --mode=644 --owner=root epd-fuse.default "/etc/default/epd-fuse"
[ ! -d "/usr/lib/systemd/system" ] && mkdir -p "/usr/lib/systemd/system" || true
install --group=root --mode=644 --owner=root epd-fuse.service "/usr/lib/systemd/system/epd-fuse.service"
make[1]: Leaving directory '/tmp/papirus/gratis/PlatformWithOS/driver-common'
Synchronizing state for epd-fuse.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d epd-fuse defaults
Executing /usr/sbin/update-rc.d epd-fuse enable
Job for epd-fuse.service failed. See 'systemctl status epd-fuse.service' and 'journalctl -xn' for details.

```

Dom1n1c, 2017-03-25:
its not an issue with the kernel right now,
the latest: https://github.com/Hexxeh/rpi-firmware/commit/41561448388d90f19f2e40ac62ce1c21c2e064ec
The latest firmware that worked before the epd fix does not work, too: https://github.com/Hexxeh/rpi-firmware/commit/52241088c1da59a359110d39c1875cda56496764

shawaj, 2017-03-25:
From Raspbian?

Just tried it from a fresh Raspbian/NOOBS and​ it worked fine.

francesco-vannini, 2017-03-25:
Tested with both a standard and the Zero W with both kernels and all is fine. Aren't you maybe trying to run the installation over an existing one?

Dom1n1c, 2017-03-25:
@francesco-vannini @shawaj 
Of course, it's not a fresh installation, but which directories do i need to clean in order to get it running?

shawaj, 2017-03-25:
Run sudo systemctl status epd-fuse.service

And post the output here please

Dom1n1c, 2017-03-25:
Hey :)

This error appears:
```Mar 25 15:25:53 ePaper epd_fuse[632]: fuse: unknown option `--panel='```

shawaj, 2017-03-25:
Can you post the full output?

shawaj, 2017-03-25:
Also can you not just start from a fresh version of Raspbian/NOOBS?

Dom1n1c, 2017-03-26:
But that's not a solution for me, there should be something, that I could revert in order to get it working (the empty panelId for instance) 

It's like rebooting a device as an act of troubleshooting. It might help for now, but no one really knows what causes the problem. 

Which directories are used by papirus and gratis?

ukscone, 2017-03-26:
I had something like that error in the early days that was caused by the panel value being unset in the file /etc/default/epd-fuse which caused the setup (systemd) to fail (and occasionally when messing around trying things)

i am doing this from memory as i don't currently have a pi with a papirus available at the moment but if you check in that file and/or run papirus-config or papirus-set I think that should fix it

Dom1n1c, 2017-03-26:
Great, that's it. 
papirus-config and disable the driver, 
papirus-set and check the screen size (2.0 for PaPiRus Zero) 
And then back again to papirus-config and enable the driver again

closed 2017-03-26 by Dom1n1c

shawaj, 2017-03-26:
Great, glad you sorted it!

@francesco-vannini shall we add this fix to an FAQ somewhere? Not sure what
causes it though

shawaj, 2017-03-26:
@ukscone has suggested this could be due to timing papirus-set without any options after it and/or installing over an existing install

shawaj, 2017-03-26:
*running, but timing