PiSupply/PaPiRus #101: Auto detect screen size and error checking

shawaj, 2017-03-26:
The ability to auto detect the screen size (negating the need for papirus-set) would be fantastic.

Additionally error checking within papirus-set, and within other scripts to make sure papirus-set had been run already would be useful

francesco-vannini, 2017-03-27:
Me and @tvoverbeek have looked into the screen signatures and unfortunately they all come up with 0x12. So unless we missed something there is no way to auto-configure the screen. We can certainly look into improving the scripts not to run if the screen hasn't been specified.
epd-fuse.default https://github.com/repaper/gratis/blob/master/PlatformWithOS/driver-common/epd-fuse.default doesn't populate EPD_SIZE on first installation.
We could either check that in each script or forcefully run papirus-set at the end of the installation process.
Either of these will solve the problem for EPD_SIZE not having a value but will not solve setting the wrong screen size which in turn causes the bar-code effect.

shawaj, 2017-03-27:
> PlatformWithOS/driver-common/epd-fuse.default doesn't populate EPD_SIZE

I guess setting the wrong size is basically impossible to avoid without it
having an identifier on the screen driver chip which it doesn't (unless
there is one but we just can't get access to it with current driver?)

Probably checking the size is set at top of scripts is best? What do you
think?

Although force running papirus-set after install is quicker to implement
and has the same effect

The only failsafe way would be to ask them on every run of the demo scripts

tvoverbeek, 2017-03-27:
The current installation script (PiSupply/PaPiRus/install on github) leaves EPD_SIZE in /etc/default/epd-fuse commented out. Hence the epd-fuse service starts with the default size 2.0.
Probably best to force the user in the papirus-setup script to define their size before starting the epd-fuse service.
In all the demo script we could add a check if environment variable EPD_SIZE is set (by sourcing /etc/default/epd-fuse). If not set prompt the user to run papirus-config to set the size and exit.

Working on these changes (and LM75B support in epd.py so the actual temperature is used in all the EPD commands).

Expect a PR in the coming days.

shawaj, 2017-03-27:
Thanks @tvoverbeek

tvoverbeek, 2017-03-29:
See PR #103

shawaj, 2017-04-14:
@francesco-vannini @tvoverbeek - seems like perhaps this is as good as we are going to be able to get it without a concrete way of detecting the screen size automatically.

@francesco-vannini - did you email Pervasive about this and if there is any other way they can be told apart from each other in code?

francesco-vannini, 2017-04-15:
Yes I got in touch but as you say this is as far as things gets on this.
We can safely close this one I suppose.

shawaj, 2017-04-15:
So they didn't reply with anything useful that could help us?

francesco-vannini, 2017-04-16:
We just got an email from Pervasive Displays.
"There is no such information embedded in the COG/driver/display.

The returned 0x12 is for G2 COG model only. No size information."

I think we can close this.

shawaj, 2017-04-16:
Cool, closing now

shawaj, 2017-04-16:
Was fixed in #103

closed 2017-04-16 by shawaj