PiSupply/PaPiRus #106: papirus-twitter does not work

Dom1n1c, 2017-04-05:
```
root@ePaper:~# papirus-twitter
/usr/local/bin/papirus-twitter: line 18: import: command not found
/usr/local/bin/papirus-twitter: line 19: import: command not found
/usr/local/bin/papirus-twitter: line 20: import: command not found
/usr/local/bin/papirus-twitter: line 21: import: command not found
/usr/local/bin/papirus-twitter: line 22: import: command not found
/usr/local/bin/papirus-twitter: line 23: import: command not found
from: can't read /var/mail/twython
from: can't read /var/mail/papirus
from: can't read /var/mail/papirus
/usr/local/bin/papirus-twitter: line 29: syntax error near unexpected token `os.path.exists'
/usr/local/bin/papirus-twitter: line 29: `if not (os.path.exists('/dev/gpiomem') and os.access('/dev/gpiomem', os.R_OK | os.W_OK)):'
```

shawaj, 2017-04-05:
Very weird. Have you made edits to your python implementation? As it is odd that it can't find the command "import" - it is a standard python method.

shawaj, 2017-04-05:
@Dom1n1c - think the shebang was broken.

Try again now

Fixed with https://github.com/PiSupply/PaPiRus/commit/069da25dd11dbc9e8e6cf5fc534b8b56bb6d062c

closed 2017-04-05 by shawaj

Dom1n1c, 2017-04-05:
Great, that's it. Is there a way to import the required modules when installing PaPiRus?
For papirus-twitter, this would be twython. 

I implemented a routine for my mqtt dashboard, to install pip and paho-mqtt

shawaj, 2017-04-05:
Do you mean you don't haven't twython installed by standard and had to install it?

On Raspbian?

Dom1n1c, 2017-04-05:
Exactly. 
It's just a suggestion to make the installation easier. 
Is there a way to check, if pip is installed or a module brought by pip? 
Installing them is possible, but it takes a few moments.

shawaj, 2017-04-05:
That's a good idea - and/or add documentation for it.

Can you submit that as a new issue?