PiSupply/PaPiRus #108: Undocumented font size attribute in text.write()

MetPhoto, 2017-04-15:
The font size attribute appears not to be documented.

The syntax appears to be text.write("Text, font_size, font_path='path to font')

Example, text in 12 point using the default font:
text.write("Hello world!", 12)

shawaj, 2017-04-15:
The font path thing was added in #98 and yes you are correct it looks like there is no readme info about it or the size attribute.

There is a font size attribute mentioned in the command line section but it also mentions there an x and y location which I'm not sure actually work either.

@francesco-vannini shall we go through all the readme API examples at some point in the next couple of weeks and make sure it's all fully and correctly documented?

shawaj, 2017-04-15:
Thanks for the heads up @MetPhoto

eat-sleep-code, 2017-06-16:
@MetPhoto How do you properly specify the font path.   I have a /fonts folder in the same folder as my python script.   It finds the fonts and works correctly if I execute the script like `python myApp.py`.   But if I try to go to the parent folder and execute `python myAppFolder/myApp.py` it can't find the font file.   I would like it to work in both scenarios.

shawaj, 2017-06-16:
Did you try it like suggested `text.write("Text, font_size, font_path='path to font')` ?

shawaj, 2017-06-16:
Actually there's a typo there. Should be `text.write("Text", font_size, font_path='path to font')

eat-sleep-code, 2017-06-16:
Sorry, wasn't clear on my question.    I am having issue with the actual font path working if i execute my python script from another folder.  It might be more of a python question (this is literally my first python project so I am learning on the go), than specific to this function.   

Basically:  

- With `python MyApp.py` the font works just fine.
- With `python MyAppFolder/MyApp.py` the script fails because it can't find the font file.

I will ideally like it to work regardless if I execute with or without the folder.

shawaj, 2017-06-16:
How have you declared the font path in the code? Can you copy and paste the line of code here?

eat-sleep-code, 2017-06-16:
I just figured it out:

`fontPath = os.path.dirname(os.path.abspath(__file__)) + /fonts/benton-sans-regular.ttf`

Then to write, I am just simply using the variable:

`text.write("Text", font_size, font_path=fontPath)`

shawaj, 2017-06-16:
Yeah I was going to say that it probably because you were using a relative path both an absolute path from root

shawaj, 2017-06-16:
@francesco-vannini perhaps we should add a demo to readme showing the font path stuff?