PiSupply/PaPiRus #109: ImportError: No module named smbus

andymule, 2017-04-18:
When using the latest Raspbian Lite installation (2017-04-10), after installing via the bash script and trying to run an example, this error is thrown:
`ImportError: No module named smbus`
This is easily resolved by running 
`sudo apt install python-smbus`
However, this should probably be handled in your installation script.

shawaj, 2017-04-18:
We don't really have the bandwidth to sorry anything other than standard Raspbian to be honest.

However I don't see any harm in adding that to the install process as if it's already installed it won't matter.

What do you think @francesco-vannini ?

francesco-vannini, 2017-04-18:
The smbus is used for the RTC part. @tvoverbeek had to create a custom version of the smbus which is available here https://github.com/PiSupply/PaPiRus/tree/master/RTC-Hat-Examples/py-smbusf the standard one should not be used.
There are detailed instructions as to what needs to be done to meet all that is needed for the RTC examples https://github.com/PiSupply/PaPiRus/tree/master/RTC-Hat-Examples
Automating the configuration needed for the RTC is however something that we should integrate including of course the smbus part.

Thanks for pointing this out.

shawaj, 2017-04-18:
@andymule - can you tell us which example you were trying to run?

shawaj, 2017-04-18:
@francesco-vannini  - looks like smbus standard version is also used.....

https://github.com/PiSupply/PaPiRus/blob/7fd84509ec15900aeb6018da9aea59c44f5b3595/papirus/lm75b.py

tvoverbeek, 2017-04-18:
The RTC on the HAT is already claimed by the kernel-module. Therefore we need the smbusf (smbus-force) module to program the reset/GPIO output going to the pogo-pin.
The LM75B is not claimed by the kernel, hence we can use of the standard smbus module for i2c access. Other examples also need additional python modules which might not be installed on raspbian-lite.

shawaj, 2017-04-18:
@tvoverbeek shall we add all dependencies to the install process do you think? that way, when already installed the system just ignores and if not installed (like on Raspbian lite perhaps) it installs them?

tvoverbeek, 2017-04-18:
Yes. We should review all examples for additional packages needed.

shawaj, 2017-04-18:
Ok great, sounds like a good idea :-)

andymule, 2017-04-20:
thanks guys, I LOVE MY PaPiRUS

shawaj, 2017-04-20:
Pleasure, thanks for reporting this :-)

closed 2017-04-22 by francesco-vannini

shawaj, 2017-04-22:
@francesco-vannini did you test this on Raspbian lite and perhaps DietPi to ensure we have all required modules?

Can't think of any others but worth testing I guess

commit be62ba60 (2017-04-22): This resolves #109 and implement #111