PiSupply/PaPiRus #111: Ability to auto-enable I2C and SPI?

shawaj, 2017-04-18:
Pimoroni have an install script method which allows them to do things like, with a single line installer, edit the config file to add lines / enable interfaces like I2C etc.

It would be great if we could perhaps get something like this up and running so that we can make it even simpler for people to install any of our products :-)

shawaj, 2017-04-18:
See here for what I'm talking about - https://github.com/pimoroni/phat-beat/blob/master/README.md

francesco-vannini, 2017-04-18:
Changing the config.txt to enable SPI and I2C should not be too difficult and could probably be done with some skillful Bash scripting.
As for a more general overarching tool that could certainly be a good idea. Let's create a new repo as soon as we have a good draft.

shawaj, 2017-04-18:
Cool beans. Did you look at the pimoroni one? Pretty neat

francesco-vannini, 2017-04-22:
This has been implemented in be62ba60382f94bf8eaceca38931893ff07fdc3b The README file has also been updated with steps now moved into the "manual" section.

closed 2017-04-22 by francesco-vannini

shawaj, 2017-04-22:
@francesco-vannini - looks awesome!

Not a major issue but just a thought - do all RasPi OSs have RasPi config?