PiSupply/PaPiRus #113: Papirus setup not working

myrdinwylt, 2017-04-25:
I recently purchased a papirus and have been trying to use it. There appears to be a problem when I setup the papirus though. After following the guide carefully the demos will not work. I am using a 2 inch papirus screen with a new raspi 0 w running rasbian jessie with pixel. When I do any demo a error like this appears.
```bash
pi@raspberrypi:~ $ sudo papirus-clock
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 131, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 74, in main
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 22, in __init__
    self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
IOError: [Errno 5] Input/output error
```
Reading the message and corresponding source files it appears to be some problem with the LM75B sensor. Please help me correct this error if you can.

myrdinwylt, 2017-04-25:
@DAlperin

myrdinwylt, 2017-04-25:
@DAlperin is working with me on this problem.

DAlperin, 2017-04-25:
Due to the nature of the problem I am starting to think it is a hardware issue. The issue persists over multiple installs of the OS and software. Does that make any sense to you guys based on your knowledge of the hardware and software? Thanks!

shawaj, 2017-04-25:
Did you turn on I2C and SPI?

What header have you soldered to the pi zero?

Is it the PaPiRus Zero?

Thanks

DAlperin, 2017-04-25:
Hey thanks for the quick response. I am forced to wait until monday until I have the hardware again. (its at a sort of local makerspace) however I know it is PaPiRus Zero, I can not confirm the other to. I am really sorry to have to ask and then not be able to work on it for a bit. You can ignore this until I post here again. Im really really sorry again... Thanks for the help!

myrdinwylt, 2017-04-25:
To answer your questions
 1. I am not aware of what 12c and SPI are
2. We have not soldered a header on but are attaching the pi and papirus with the stuctaral pins included in the box in away that allows the stacking headers to touch the pi. We are using generic stacking headers.
3. We are using the papyrus 0 2.0 inch

DAlperin, 2017-04-25:
@myrdinwylt I think I enabled I2c and SPI. Also we are in fact using the PaPiRus Zero. Lets wait on this for now until we have access to the pi again. Thanks!

francesco-vannini, 2017-04-26:
The errors you have @myrdinwylt are because the I2C and SPI aren't enabled. There is a section in the README.md which tells you how to manually enable I2C and SPI. This has however been automated in the latest commit.
For a brief time there was an issue with smbus not being installed automatically. This was needed by LM75B and it has now been included in the auto installation. In case you are not on the latest branch @stocktoncycles please run sudo apt-get install python-smbus which should fix your problems.

If you get confused what I suggest is to use a spare SD, flash the latest Raspbian and run the auto configuration once again. 
The post from our good friend Frederick is now outdated and issues about sudo are no longer current. Please disregard the majority of that article as the issues have now been fixed in the current branch.

stocktoncycles, 2017-04-26:
@francesco-vannini , I'm sure it's possible that I2C and SPI we're not enabled though I did manually enable them in the config tool before re-starting the pi. I will check to see if smbus is installed. Thanks.

francesco-vannini, 2017-04-28:
@stocktoncycles and  @myrdinwylt can you confirm whether you managed to make any progress on this? I don't think this is an issue though. I'll keep this open for a little while to give you time to get back.

closed 2017-04-29 by francesco-vannini

DAlperin, 2017-04-29:
Thank you guys for the help we will try you solutions on Monday and they all sound like they will work. Thanks!

vanillabrand, 2017-05-15:
Hi i have the same problem. i've installed as instructed and enabled I2C & SPI but it's still showing me the same error. Is the error because it can't communicate with the screen? Help!

shawaj, 2017-05-15:
@vanillabrand what is the error you are getting?

vanillabrand, 2017-05-15:
Thanks @shawaj  
`pi@raspberrypi:~ $ sudo papirus-clock
Traceback (most recent call last):
  File "/usr/local/bin/papirus-clock", line 131, in <module>
    main(sys.argv[1:])
  File "/usr/local/bin/papirus-clock", line 74, in main
    papirus = Papirus()
  File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
    self._lm75b = LM75B()
  File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 22, in __init__
    self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
IOError: [Errno 5] Input/output error`

I've tried installing manually too. Enabled SPI & I2C, installed everything using sudo but it still shows the same error. Any clues?

thanks!

shawaj, 2017-05-15:
What's the result if you run `sudo i2cdetect -y 1` ?

vanillabrand, 2017-05-15:
`
           0    1   2   3   4   5   6   7  8   9   a   b   c   d   e   f
00:                 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10:       -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20:       -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30:       -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40:       -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50:       -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60:       -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70:       -- -- -- -- -- -- -- -- 
`

francesco-vannini, 2017-05-15:
Can you please try on a different Raspberry Pi? Preferably one with a pre-soldered header. Which header did you use for your Pi Zero?

vanillabrand, 2017-06-13:
Ah ok found the screen wasn't working. Got a new one on order. Thx

shawaj, 2017-06-13:
@vanillabrand - the screen not working is very unlikely. We have had very few go wrong as they're tested in the factory.

What header are you using on the Pi Zero?

vanillabrand, 2017-06-14:
It's pretty likely if it had a hairline crack running through it. New one
arrived and works a treat ;) thx

shawaj, 2017-06-14:
@vanillabrand fantastic news!