PiSupply/PaPiRus #115: Papirus wont work on new Raspberry Pi Zero W

extremepilot, 2017-04-27:
I recently purchased the new wireless Raspberry Pi Zero and tried to use my Papirus screen with it, but it will not work and I get one of a number of errors, images attached. I have confirmed the display works on the regular Pi Zero and the Pi 3-B.

[image]
[image]

shawaj, 2017-04-27:
Are you using Raspbian?

Have you tried with a fresh image and got it fully updated first?

Thanks

extremepilot, 2017-04-27:
Yeah, I have updated everything multiple times. I have also tried both Raspbian and Raspbian lite.

shawaj, 2017-04-27:
@francesco-vannini any ideas here? I've not seen any issues with the Raspberry Pi Zero Wireless have you?

@extremepilot what header have you soldered to the Pi?

francesco-vannini, 2017-04-28:
There are no known problems with any of the Raspberry Pis. Have tested for the last PR with a Zero W too. If you have correctly enabled SPI and I2C (which are now enabled by default from our script anyway) I suspect something dodgy is happening with your Zero @extremepilot . I'll keep this open for a while so that you can get back to us on how this goes.

extremepilot, 2017-04-28:
Issue seems to be resolved on Raspbian, will confirm for Raspbian lite soon.

shawaj, 2017-04-28:
@extremepilot perfect thanks

extremepilot, 2017-04-28:
Still having issues with Raspbian lite, with regular Raspbian the issue seemed to exist with NOOBS 2.3 and was resolved with NOOBS 2.4.

francesco-vannini, 2017-04-28:
For Raspbian lite I have used the April release and all is ok. What error do you get now? Are you using the April release and flash directly on SD?

extremepilot, 2017-04-28:
I am using the most up to date version of NOOBS (2.4, 4/10) to install Raspbian lite onto the microSD card.
[image]

shawaj, 2017-04-28:
Looks like smbus hasn't installed properly? 

Thought we added this to the install script @francesco-vannini ?

extremepilot, 2017-04-29:
The issue seems to be solved on both instances of Raspbian. Sorry for the confusion, there are a lot of steps to setup PaPiRus that one can get lost in. The final issue for me seemed to be enabling i2c and SPI.

francesco-vannini, 2017-04-29:
I am glad it all works now but I fail to see the many steps involved in setting up the PaPiRus as all is done now via the one liner at the top of the README file.
Running
curl -sSL https://pisupp.ly/papiruscode | sudo bash
should do everything you need.
Did you by any chance follow the manual setup instead for some reason? Because that wasn't clear to me.
Anyways all good in the end, I'll go ahead and close the issue.

closed 2017-04-29 by francesco-vannini