PiSupply/PaPiRus #118: UTF characters do not display with PapirusTextPos()

tsorman, 2017-05-02:
When I am trying to add a UTF character when using the PapirusTextPos()
I get a decoding error. 

The code is: 
from papirus import PapirusTextPos
text = PapirusTextPos()
text.AddText( u"\u304b" )

and the error is : 

Traceback (most recent call last):
  File "sizedFont.py", line 3, in <module>
    text.AddText( u"\u304b" )
  File "/usr/local/lib/python2.7/dist-packages/papirus/textpos.py", line 38, in AddText
    self.allText[Id] = DispText(text.decode('string_escape'), x, y, size)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u304b' in position 0: ordinal not in range(128)

I tried changing the string to "Hello" and it worked fine.

Note that I am not that strong in Python so I might be doing something wrong here but the error appears to be on the textpos.py

Since the PapirusText() works fine with UTF characters, ( I managed to display some Japanese text) 
then I should think that the behavior between the two classes should be the same.

tsorman, 2017-05-02:
Well I managed to get around the problem by accessing textpos.py on the library and removing the .decode('string_escape')

Again Python is not my strong point but if that is not affecting the performance, should it be on the next update?

francesco-vannini, 2017-05-02:
For some reason you have a previous version of textpos. You should pull the latest code and reinstall, the textpos file has been amended as shown here https://github.com/PiSupply/PaPiRus/pull/112/files

closed 2017-05-02 by francesco-vannini