PiSupply/PaPiRus #120: Free UART in next version of Papirus Zero HAT

arofarn, 2017-05-20:
The UART pins (8->TX and 10->RX) are used by Papirus HAT for border control and discharge in the current version of the hardware.

It seems that there is no reason that Papirus cannot use others pin for border control discharge (like pins 13 and 15 or 29 an 31 may be?). But it is difficult to use software serial on pins left free by the HAT.

Please, free GPIO pin 8 and 10 in the next version of Papirus.

francesco-vannini, 2017-05-23:
Hi @arofarn that's a good point you are making. We have no plans at the moment to issue a new design for the board but we'll certainly keep this in mind should there be a new version.
Also frankly I don't know which were the reasons for which our engineers choose those pins instead of others but I suspect it wasn't done by chance. We'll check with them anyway.

arofarn, 2017-05-23:
OK, thank you!

shawaj, 2017-06-12:
@francesco-vannini did you ever find out about this? should we leave it open as a potential future hardware improvement?

and what about we add a "hardware" label on all out github pages for hardware questions?

shawaj, 2017-08-09:
See also https://github.com/PiSupply/PaPiRus/issues/25