PiSupply/PaPiRus #121: PaPiRus not rendering screen

tomaso909, 2017-05-26:
I am using the 2.7 screen Papirus on a PI 3. I had it working for a while but after an apt-get upgrade, it stopped working. I also fiddled a bit with the PI settings because I was trying to get a GPS connected with bit-banging on ports 22 and 4 to work, but I believe I have now set everything as it was before. Further, I reinstalled the drivers of this repo but it still does not work.

I do not get any error but the screen does not show anything. For example, running `papirus-clock` writes `panel = EPD 2.7 264 X 176 version=4 COG=2 FILM=231` in the terminal but the screen is empty.

Is there any tests I can perform? Any ideas are welcomed.

francesco-vannini, 2017-05-26:
First of all I would recommend getting a spare SD card and install a fresh
Raspbian on it. Verify that all is well with PaPiRus then maybe use that
system as a baseline to understand what you changed on your current one
that made the board not work as it should.

shawaj, 2017-06-06:
@tomaso909 did you manage to sort this?

tomaso909, 2017-06-07:
Yes. It worked after a fresh install.

closed 2017-06-07 by francesco-vannini