PiSupply/PaPiRus #122: No module named 'lm75b' on latest build of Raspbian Lite with Python 3

eat-sleep-code, 2017-05-29:
I used the following command to install PaPiRus on my Pi that is running Raspbian Lite with Python 3.4.2

`curl -sSL https://pisupp.ly/papiruscode | sudo bash`

I then put the following in my python file

`from papirus import PapirusText`

I then run my code using python3 and receive an error of:

ImportError: No module named 'papirus'

So, then I try running:

`sudo python3 setup.py install`

It throws an error on line 33 of `/usr/local/lib/python3.4/dist-packages/papirus/lm75b.py` about invalid syntax.

How do I get this working.

shawaj, 2017-05-29:
This does not currently work with python 3. Please use python 2.7 for now

eat-sleep-code, 2017-05-29:
Time to move along folks.   Python 2.7 is no longer being maintained.  Python 3.0 was released back in 2008.  

Do we have an ETA on when 3.0 is going to be supported?

shawaj, 2017-05-29:
Nope we do not currently have any plans for python 3 support. The default for Debian is still 2.7

eat-sleep-code, 2017-05-29:
Default or not, Python 2 drops out of support in 2.5 years so "not currently have any plans for python 3" seems kind of silly.  

Kind of makes me rethink buying items from piSupply.

Luckily, it appears my code works okay in Python 2.

shawaj, 2017-05-29:
Debian don't seem to concerned about python 2.7 no longer being maintained and they would have much much larger reasons and a longer process to change so we will probably do it after they announce a change.

For the time being, it's more important to support the default python version in our opinion. 

Having said that, we'd love to be able to support both it's just not the top of our task list right now. But we would definitely accept a PR for a python version agnostic codebase if you're interested in helping us ;-)

shawaj, 2017-05-29:
*too

francesco-vannini, 2017-05-30:
Why would that make you rethink buying for Pi Supply? This is the kind of
"white" threat that is really miss placed and doesn't actually encourage
anybody helping you. Proactive attitude and an overall nicer approach are
much more welcome for the future.

jas-per, 2017-06-04:
seriously, python2 means legacy code in 2017, using py3 is the only sane choice for any new project started these days! I didn't explicitly check python3 support before buying the papirus and am very frustrated about this .. py3s asyncio would be a perfect fit for interrupt handling/buttons and background tasks like full redraws of the panel

debians struggle with replacing their numerous legacy py2 scripts is no excuse at all for a tiny lib to not support python3 in 2017: at least you should explicitly state **python2** support for your product instead of general python support. and btw, the next debian (buster) is going to default to py3, exactly because py2 support is going to end

sorry for the rant, might come as a suprise if you're not that much into using python day by day, but since you're offering a commercial product you should imho be serious about your engagement - maybe offer some free test kits as bounty for people doing this job for you or whatever..

shawaj, 2017-06-04:
@jas-per as said above, when the Raspbian default changes to python 3 so will we. Until then, we are too small a team to fix something that isn't broken. 

You don't need to tell us about the development of Debian, Raspbian and the like - we are well aware of that and we have a valid reason not to change yet (whether you agree with our reason or not is a different thing altogether and not an argument we care to have with you). We have never explicitly stated python 3 support anywhere and since it's not the default python yet, on Raspbian you should probably assume no python 3 support unless stated otherwise.

Having said that, if someone wants to help us with python 3 support at this stage we're willing to accept PRs and we do offer test gear to developers that are keen to help however as previously mentioned we are a small team and as such we currently have no formal process for this.

shawaj, 2017-06-04:
*willing to accept PRs as long as it doesn't break backwards compatibility with the current default that is

jas-per, 2017-06-04:
thanks for the prompt answer! ;) don't want to argue about what "python" means in 2017, there are already more than enough discussions about this topic all over the internet. no python3 support **does** come as a surprise tough. don't want to tell you about the limited possibilities of a small team either - I'm quite aware of the fact that everything that works fine if its all about software gets more difficult if hardware/commercial interests become involved

I think you're offering some great hardware, so I would not have a problem in spending some time improving the python bindings, but I'd have to be sure that your team is dedicated about the future availability of your product. being fine with python2 only these days doesn't convey a good attitude in this regard, its more like "what do we care about what happens in a year from now"

in regard to no formal process: being more open/explicit about your intention to strengthen the community of this project by sharing test gear and having some tags/labels in the issue tracker about issues that you deem important for improvements might be a start?

shawaj, 2017-06-04:
@jas-per - as said we don't want to discuss the merits of python 3 vs python 2. Please don't just come in here and tell us how we should or shouldn't run our business and presume that because we disagree with you about python versions that we don't properly support our products. It is an incredibly patronising view, and not a great way to start an open discussion. 

Just take a look for a minute at the activity on this repo over the past year - PRs, issues etc - every one responded to, sorted and lots of work going on both from our staff and the lovely people that volunteer their efforts too.

We have our reasons for not wanting to do python 3 support yet, which we've already explained to you - python 2 works fine for the purposes of PaPiRus, the vast majority of our customers don't care which it uses, python 2 is the default now and even when python3 is the default it won't break overnight. We have a small team and we will move to python 3 whenever there is a need, which there really isn't right now. You may disagree with that, but that's our position so please respect it.

PaPiRus isn't going anywhere soon. We are accept designing more PaPiRus boards as we speak for BeagleBone and other platforms and larger and colour ePaper displays.

It is great to hear you would be happy to help with the codebase, and we really do appreciate it - we already have a huge number of volunteer contributions and feature suggestions but the more the better as far as we are concerned!

Take a look at the issues and get cracking :-)

shawaj, 2017-06-04:
*actually, not accept

jas-per, 2017-06-04:
lesson learned - always check your foundations first ;) since all this lib does is writing images to /dev/epd converting to python3 was trivial: replace PIL with pillow (last release of PIL was 10 years ago btw) and change all reads/writes to /dev/epd to explicitly use bytes. not too complicated to turn into a py2/3 compatible codebase, can't be bothered tough, please do your homework at some stage 

@shawaj sorry about not having an open discussion, but in the end: if you'd put some effort into your codebase instead of justifying your ignorance, this discussion would not have been necessary

Take a look at the issues and get cracking :-)

shawaj, 2017-06-04:
@jas-per not quite sure why there is a need for such hostility from you. 

We are perfectly capable of changing to python 3 but we have decided not to. Disagreeing with you over priorities doesn't mean we are ignorant or incapable... Just that we disagree with you. 

It's sad to hear that you're not willing to contribute back to the codebase but thanks for commenting anyway. If you change your mind you know where we are.

tvoverbeek, 2017-06-04:
Just to chime in. I have had adaptation to python3 on my todo list for
PaPiRus for some time.
Yes, the base code is going to be relatively easy to convert so it runs on
both python2 and python3.
A bit more complicated is the recently added unicode support. There are
quite some differences how unicode
is treated in python2 and python3.
However I will not be able to do very much before July (Going on a long
trip to our relatives down under).
Goal is to have the same code running on both python2 and python3 using
version checking in the code.

shawaj, 2017-06-06:
@tvoverbeek thanks :-)

closed 2017-07-02 by eat-sleep-code

shawaj, 2017-07-03:
@eat-sleep-code did you work out what you needed to here then?

CharlesMougenot, 2017-07-31:
Hi Everyone, I really like the papirus screen but I really need to get it work with python 3 to get it work with my robotic project. Do you plan to make papirus working on python 3? Thanks in advance for your help :-)

shawaj, 2017-07-31:
It's on it to do list for future but not sure when it'll be done.

Thanks

CharlesMougenot, 2017-07-31:
Ok Thanks