PiSupply/PaPiRus #123: Sanity check test as part of install process

projectarc-co-uk, 2017-06-06:
This is a feature request that arose from a recent discussion with Aaron.

While installing a PaPiRus Zero on a Zero 1.3 I kept hitting an ErrNo 103 which it appears a number of people have seen in the past, along with various other ErrNos (2, 107, etc). There isn't much in the way of error log/messages to help diagnose what the core problem is in these cases.

In my case it turned out that one of the header pins on the Pi was about 1mm too low, so it wasn't connecting properly. This was after 4 or 5 hours of trying various reinstall and reconfig approaches, until I eventually tried again on a spare Pi.

So my request is for some kind of sanity check/test that can be run as part of the install/config process, which could determine which pieces/layers are working and what isn't. For example, check whether SPI bus reports the display is attached, use i2c-tools or similar to see if the i2c layer reports the display as present and with the right id, check the GPIO layer to see if the display board is responding correctly, and also check the epd/fuse driver & service to see if it is all responding as expected.

This would make it much easier to diagnose any problems during initial setup, when frustrations are most likely to occur. In my case, it would have been really handy to know the i2c, spi, epd, fuse, python stuff was all fine, but the GPIO pins weren't all attached.

I have no idea how much work this would be, but I'd guess it's not a trivial script to write, so feel free to throw this on the wishlist pile ;-)

shawaj, 2017-06-07:
Similar to #114 and #72 in thoughts. Perhaps these should all be combined when eventually fixed.

shawaj, 2017-06-07:
@projectarc-co-uk thanks for this :-)