PiSupply/PaPiRus #124: IOError: [Errno 121] Remote I/O error

eat-sleep-code, 2017-06-07:
I have a new Pi Zero W with the PaPiRus Zero.   When I run any of the command line items, I get an error similar to: 

> File "/usr/local/bin/papirus-clear", line 29 in <module> 
> main()
> File "/usr/local/bin/papirus-clear", line 23 in main
> papirus = Papirus()
> File "/usr/local/lib/python2.7/dist-packages/papirus/epd.py", line 58, in __init__
> self._lm75b = LM75B()
> File "/usr/local/lib/python2.7/dist-packages/papirus/lm75b.py", line 22, in __init__
> self._bus.write_byte_data(self._address, LM75B_CONF_REGISTER, LM75B_CONF_NORMAL)
> IOError: [Errno 121] Remote I/O error

The Pi Zero is running the latest version of Jessie Lite.

tvoverbeek, 2017-06-07:
Smells like the Zero W cannot communicate with the lm75b temperature sensor via I2C.
Check I2C is enabled with raspi-config.
If enabled then use i2cdetect (from the i2c-tools package, install if not there) .
'i2cdetect -y 1' should report a device on address 0x48 (the lm75b). If not check the connections between the Zero W and Papirus Zero.

shawaj, 2017-06-13:
@eat-sleep-code did you get this sorted?

eat-sleep-code, 2017-06-16:
Yeah, apparently a bad solder job.   Went to hammerless headers and it is working.

closed 2017-06-16 by eat-sleep-code

shawaj, 2017-06-16:
@eat-sleep-code you mean that you're now using the hammer header or were before? 

Because we've had issues with the hammer headers not getting a good connection previously. We recommend everyone to solder or buy one of our pre soldered Pi Zeros.

In any case, glad you've got it working

neuberfran, 2019-04-12:
[image]
anyone

MichaelSternNYC, 2019-05-11:
I was getting the same error message so, per the advice above, I revisited my solder joints and cleaned up anything that looked suspicious. I also took he opportunity to remove a small splatter of solder on the underside of the board. Victory! Test code is now running without error. The hardware does as it should; when I power the machine down, my test remains on the screen.

shawaj, 2019-05-11:
@MichaelSternNYC thanks for the input. What was the solder spatter and what were the dodgy looking connections. Would be interesting to know for our reference and to improve manufacture quality

MichaelSternNYC, 2019-05-11:
Not your lousy soldering — mine. I hadn’t done a great job putting headers on my Pi.

shawaj, 2019-05-11:
Aha! Even better news 😁